acados/acados #837: MEX cannot find library 'acados'

T3AR5, 2022-07-27:
Hi,

I have installed Acados on my Windows PC using Matlab and want to use it for my master thesis.

Following the instructions for the installation I get an error using MEX, because the lib folder is empty except for a textfile "link_libs.json".

> Error using mex
> MEX cannot find library 'acados', specified with the -l option.
>  MEX searched for a file with one of the following names:
>  libacados.a
> acados.lib
> libacados.lib
>  Verify the library name is correct. If the library is not
>  on the existing path, specify the path with the -L option.
> 
> 
> Error in ocp_compile_interface (line 169)
>         mex(mex_flags, FLAGS, LDFLAGS, COMPDEFINES, COMPFLAGS, acados_include, acados_interfaces_include, external_include, blasfeo_include, hpipm_include,...
> 
> Error in acados_ocp (line 181)
>                 ocp_compile_interface(obj.opts_struct);
> 
> Error in minimal_example_ocp (line 108)
> ocp = acados_ocp(ocp_model, ocp_opts);

I already reinstalled three times, following the different instructions supplied with the build, but still the lib folder remains empty. I also looked at the other threads, but couldnt find a solution that worked for me.

What do I need to change for it to be created correctly? Is it possible to get the files directly and copy paste them into the folder? Where would I find the files?

Thanks for your help and best regards

T3AR5, 2022-07-27:
.

FreyJo, 2022-07-28:
Hi,

I don't know why setting up the environment variable `ACADOS_INSTALL_DIR` does not work for you.
But this is the problem.
In your CMakeCache it says:
```
ACADOS_INSTALL_DIR:PATH=C:/Users/Gabriel/acados/build/$ACADOS_INSTALL_DIR
```

Please try to set it to
```
C:/Users/Gabriel/acados/lib
```
directly in CMake by putting the hard coded path there instead of the environment variable.

closed 2022-08-03 by FreyJo