acados/acados #838: Using DNN with acados

sdfzz, 2022-07-27:
Dear all,

I'm trying to use acados for my research. For my study, I need to replace the dynamic model of the system with a deep neural network. 

So, my AcadosModel.f_expl_expr needs to be a DNN network....however, since acados uses casadi, I cannot seem to directly use DNN as  AcadosModel.f_expl_expr.

Is such setup possible with acados? With popularity of deep learning, I'm sure many people want to use DNN with acados.

Any help would be appreciated

Regards,

Steve

FreyJo, 2022-07-28:
Hi,

this repository might be interesting for you:
https://github.com/TUM-AAS/ml-casadi

It was open-sourced very recently.
And I am personally also interested to check it out more and to get to know what are the limitations of this workflow.
I'd be happy to hear what you find.

Cheers!

sdfzz, 2022-07-29:
Hi,

Thank you for the link. I will check out the code + paper

I will let you know if there's any update

Regards,

Steve

sdfzz, 2022-08-09:
Dear all,

It seems that one way to include DNN into ACADOS is to setup DNN using CasADi...

ex) https://ieeexplore.ieee.org/abstract/document/9834096

I succeeded in setting up a CasADi-based DNN. However, 'AcadosOcpSolver' takes forever to build (about 30 min)

Normally, without DNN, building AcadosOcpSolver takes less than a second. I think by adding DNN, the system gets too complex...

I'm running ACADOS in Python3.

So, the question is, is there any method to 'save' my ocp solver and load it again? Then, I don't have to wait 30 minutes everytime I'm running a simulation

Regards,

Steve