acados/acados #842: Is it possible considering the implement for Apple M1 chip

tomcattiger1230, 2022-08-05:
Hi,

has anyone tried to implement and test on Apple M1 chip?

FreyJo, 2022-08-05:
Hi,
acados has been tested on MacOS with the MacBookPro version just before the M1 chip.
Plus there is a dedicated BLASFEO target for M1, which still should be added to CMake though, https://github.com/giaf/blasfeo/blob/master/Changelog.txt#L57
But with `GENERIC` BLASFEO target it should work fine.

tomcattiger1230, 2022-08-05:
> Hi,
> acados has been tested on MacOS with the MacBookPro version just before the M1 chip.
> Plus there is a dedicated BLASFEO target for M1, which still should be added to CMake though, https://github.com/giaf/blasfeo/blob/master/Changelog.txt#L57
> But with `GENERIC` BLASFEO target it should work fine.

Hi, I've tested it with MacBook Pro Intel-Version. But now it faces arm64 Probem (no work with x86_64)

tomcattiger1230, 2022-08-06:
Hi, 

clear to say, if one uses C++ it is ok. But if one uses Python, it comes with error 
ld: warning: ignoring file /Users/weiluo/Develop/github/acados/lib/libacados.dylib, building for macOS-x86_64 but attempting to link with file built for macOS-arm64

tomcattiger1230, 2022-08-06:
Hi, 

I've noticed that it is caused by ctypes.CDLL
a = CDLL('/Users/xxx/Develop/github/acados/lib/libacados.dylib')

=>.  OSError: dlopen(/Users/xxx/Develop/github/acados/lib/libacados.dylib, 0x0006): tried: ''/Users/xxx/Develop/github/acados/lib/libacados.dylib' (mach-o file, but is an incompatible architecture (have (arm64), need (x86_64)))

FreyJo, 2022-08-08:
It seems that the Python you are using is `x86_64` emulated, but the `dylib` is compiled for `arm64` and thus not compatible.
I think you could try to either make the Python native `arm64` or compile the `dylib` for `x86_64`.

tomcattiger1230, 2022-08-08:
> It seems that the Python you are using is `x86_64` emulated, but the `dylib` is compiled for `arm64` and thus not compatible.
> I think you could try to either make the Python native `arm64` or compile the `dylib` for `x86_64`.

Thanks. I've found the problem. It works now.

closed 2022-08-08 by tomcattiger1230

FreyJo, 2022-08-08:
Could you explain, how you solved it here, in case someone else has this issue?

tomcattiger1230, 2022-08-08:
> 

I use anaconda to set up Python env. But the default download from anaconda for macOS currently is still for x64_86, and I've not noticed about that. One can choose manually the version for arm64. After reinstalling anaconda for arm64, this library problem is solved.