acados/acados #899: Unable to install on Windows

ritvikmahajan01, 2022-12-09:
I'm following the installation instructions to install acados on Windows 11, but when I run the following command:
`mingw32-make.exe -j4`
I am getting this error message in the powershell:
```
D:\acados\acados\utils\timing.c: In function 'acados_tic':
D:\acados\acados\utils\timing.c:115:36: warning: implicit declaration of function 'clock_gettime' [-Wimplicit-function-declaration]
 void acados_tic(acados_timer* t) { clock_gettime(CLOCK_MONOTONIC, &t->tic); }
                                    ^~~~~
D:\acados\acados\utils\timing.c:115:50: error: 'CLOCK_MONOTONIC' undeclared (first use in this function)
 void acados_tic(acados_timer* t) { clock_gettime(CLOCK_MONOTONIC, &t->tic); }
                                                  ^~~~~~~
D:\acados\acados\utils\timing.c:115:50: note: each undeclared identifier is reported only once for each function it appears in
D:\acados\acados\utils\timing.c: In function 'acados_toc':
D:\acados\acados\utils\timing.c:121:19: error: 'CLOCK_MONOTONIC' undeclared (first use in this function)
     clock_gettime(CLOCK_MONOTONIC, &t->toc);
                   ^~~~~~~
acados\CMakeFiles\acados.dir\build.make:660: recipe for target 'acados/CMakeFiles/acados.dir/utils/timing.c.obj' failed
mingw32-make[2]: * [acados/CMakeFiles/acados.dir/utils/timing.c.obj] Error 1
mingw32-make[2]: * Waiting for unfinished jobs....
CMakeFiles\Makefile2:360: recipe for target 'acados/CMakeFiles/acados.dir/all' failed
mingw32-make[1]: * [acados/CMakeFiles/acados.dir/all] Error 2
Makefile:144: recipe for target 'all' failed
mingw32-make: * [all] Error 2
```

As suggested online, I have tried adding the `-D_POSIX_C_SOURCE=199309L` switch in the cmake command (before the `mingw32-make.exe -j4 command`) but that doesn't seem to work.

Can someone please suggest how can I fix this?

FreyJo, 2022-12-11:
Hi,
did you check this issue?
https://github.com/acados/acados/issues/798

ritvikmahajan01, 2022-12-11:
Thanks for the reply. Adding the `-DCMAKE_BUILD_TYPE=Debug` in the cmake command fixes the error which was earlier shown in the powershell (I had earlier tried adding `-D_POSIX_C_SOURCE=199309L` in the cmake command but that doesn't seem to have any effect). However, now when I'm trying to run the `minimal_example_ocp`, I'm getting this error in Matlab:
```
compiling D:\acados\examples\acados_matlab_octave\..\..\interfaces\acados_matlab_octave\ocp_create.c
Building with 'MinGW64 Compiler (C)'.
Error using mex
C:/ProgramData/MATLAB/SupportPackages/R2022a/3P.instrset/mingw_w64.instrset/bin/../lib/gcc/x86_64-w64-mingw32/6.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe:
skipping incompatible D:\acados\examples\acados_matlab_octave\..\..\lib\acados.lib when searching for -lacados
C:/ProgramData/MATLAB/SupportPackages/R2022a/3P.instrset/mingw_w64.instrset/bin/../lib/gcc/x86_64-w64-mingw32/6.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe:
cannot find -lacados
C:/ProgramData/MATLAB/SupportPackages/R2022a/3P.instrset/mingw_w64.instrset/bin/../lib/gcc/x86_64-w64-mingw32/6.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe:
skipping incompatible D:\acados\examples\acados_matlab_octave\..\..\lib\hpipm.lib when searching for -lhpipm
C:/ProgramData/MATLAB/SupportPackages/R2022a/3P.instrset/mingw_w64.instrset/bin/../lib/gcc/x86_64-w64-mingw32/6.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe:
cannot find -lhpipm
C:/ProgramData/MATLAB/SupportPackages/R2022a/3P.instrset/mingw_w64.instrset/bin/../lib/gcc/x86_64-w64-mingw32/6.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe:
skipping incompatible D:\acados\examples\acados_matlab_octave\..\..\lib\blasfeo.lib when searching for -lblasfeo
C:/ProgramData/MATLAB/SupportPackages/R2022a/3P.instrset/mingw_w64.instrset/bin/../lib/gcc/x86_64-w64-mingw32/6.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe:
cannot find -lblasfeo
C:/ProgramData/MATLAB/SupportPackages/R2022a/3P.instrset/mingw_w64.instrset/bin/../lib/gcc/x86_64-w64-mingw32/6.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe:
skipping incompatible D:\acados\examples\acados_matlab_octave\..\..\lib\osqp.lib when searching for -losqp
C:/ProgramData/MATLAB/SupportPackages/R2022a/3P.instrset/mingw_w64.instrset/bin/../lib/gcc/x86_64-w64-mingw32/6.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe:
cannot find -losqp
collect2.exe: error: ld returned 1 exit status

Error in ocp_compile_interface (line 175)
        mex(mex_flags, FLAGS, LDFLAGS, COMPDEFINES, COMPFLAGS, acados_include, acados_interfaces_include, external_include, blasfeo_include, hpipm_include,...

Error in acados_ocp (line 181)
                ocp_compile_interface(obj.opts_struct);

Error in minimal_example_ocp (line 108)
ocp = acados_ocp(ocp_model, ocp_opts);
```

FreyJo, 2022-12-12:
Hi,
```
skipping incompatible D:\acados\examples\acados_matlab_octave\..\..\lib\blasfeo.lib when searching for -lblasfeo
```
This makes me think that the `.lib` files are probably not compiled with the same mingw version that Matlab is trying to use.

Maybe you can verify that.
Alternatively, you can try to use this branch, where a workflow to compile all of acados from Matlab was added:
https://github.com/acados/acados/pull/897

ritvikmahajan01, 2022-12-15:
Thanks for sharing the modified branch. Unfortunately, I'm encountering some errors with this also. When I run the acados_install.m file, I get this:

```
>> acados_install
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/ProgramData/MATLAB/SupportPackages/R2022a/3P.instrset/mingw_w64.instrset/bin/gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/ProgramData/MATLAB/SupportPackages/R2022a/3P.instrset/mingw_w64.instrset/bin/c++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type is Release
-- ACADOS_WITH_OPENMP: OFF
CMake Error at external/CMakeLists.txt:32 (add_subdirectory):
  The source directory

    D:/acados/external/blasfeo

  does not contain a CMakeLists.txt file.

CMake Error at external/CMakeLists.txt:41 (add_subdirectory):
  The source directory

    D:/acados/external/hpipm

  does not contain a CMakeLists.txt file.

--  
-- Target: BLASFEO is X64_AUTOMATIC, HPIPM is X64_AUTOMATIC
-- Linear algebra: HIGH_PERFORMANCE
-- Matlab MEX (OFF)
-- Octave MEX (OFF)
-- Octave Templates (OFF)
-- System name:version Windows:10.0.22623
-- Build type is Release
-- Installation directory is D:/acados
-- OpenMP parallelization is OFF
--  
-- Configuring incomplete, errors occurred!
See also "D:/acados/build/CMakeFiles/CMakeOutput.log".
Error using acados_install
cmake command failed:
cmake.exe -G "MinGW
Makefiles"
-DACADOS_INSTALL_DIR=%ACADOS_INSTALL_DIR%
-DBUILD_SHARED_LIBS=OFF
-DACADOS_WITH_OSQP=OFF ..
```

FreyJo, 2022-12-17:
Did you pull the acados submodules as described?

Ajin2305, 2023-11-13:
Hi @FreyJo  

I encountered a similar issue (Not being able to identify clock_gettime()) when running Acados on a Windows machine with a MinGW compiler.  I did some research and understood the following.

POSIX defines the function clock_gettime(), and Windows is not POSIX compliant. [https://stackoverflow.com/questions/57668563/clock-gettime-identifier-not-found-in-visual-studio-in-windows-10](url)

I did however find that QueryPerformanceCounter could still be utilized by windows (Win10 at least). This traditionally has a resolution of 15.625ms, which can be modified using timeBeginPeriod function. 
[https://stackoverflow.com/questions/31323214/how-to-coordinate-threads-properly-based-on-a-fixed-cycle-frequency](url)

Modifying the function in timing.c so as to include QueryPerformanceCounter for any Windows Machine should fix this issue.
If it comes to accurate profiling (up to 1ms) we can do it using the function timeBeingPeriod as mentioned above.