acados/acados #901: Unable to compile in Matlab interface on macOS M1

Mathiasschuler96, 2022-12-20:
Hi 

I was able to install ACADOS but when I try to run an example I get several of the following errors:

...building for macOS-x86_64 but attempting to link with file built for macOS-arm64...

Do you have any advice on how to solve this issue?

Thank you in advance!

FreyJo, 2022-12-20:
Hi,

can you check this thread?
The issue isn't fully sovled there and compiling for Matlab and M1 seems tricky, but maybe you can get further!
https://discourse.acados.org/t/matlab-crash-illegal-instruction-on-m1-mac/865/7

Mathiasschuler96, 2022-12-20:
I've changed the BLASFEO_TARGET in the CMakeLists.txt file to "GENERIC" as written in the thread but the error is still the same unfortunately...

FreyJo, 2022-12-20:
Did you also check this thread?
https://discourse.acados.org/t/error-using-mex-undefined-symbols-for-architecture-x86-64/830

Mathiasschuler96, 2022-12-20:
Yes I did but I do not understand how to "compile the acados library also for the emulated x86_64 target."

Mathiasschuler96, 2022-12-20:
I tried to reinstall acados with 
```
mkdir -p build
cd build
cmake -DACADOS_WITH_QPOASES=ON ..
# add more optional arguments e.g. -DACADOS_WITH_OSQP=OFF/ON -DACADOS_INSTALL_DIR=<path_to_acados_installation_folder> above
make install -j4
```
with the BLASFEO_TARGET in the CMakeLists.txt file set to "GENERIC" but got the following CMakeError.log:

```
Detecting C compiler ABI info failed to compile with the following output:
Change Dir: /Users/user/acados/build/CMakeFiles/CMakeScratch/TryCompile-ywRMou

Run Build Command(s):/opt/homebrew/bin/gmake -f Makefile cmTC_2e99a/fast && /opt/homebrew/bin/gmake  -f CMakeFiles/cmTC_2e99a.dir/build.make CMakeFiles/cmTC_2e99a.dir/build
gmake[1]: Entering directory '/Users/user/acados/build/CMakeFiles/CMakeScratch/TryCompile-ywRMou'
Building C object CMakeFiles/cmTC_2e99a.dir/CMakeCCompilerABI.c.o
/Library/Developer/CommandLineTools/usr/bin/cc   -arch “x86_64” -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.1.sdk   -v -Wl,-v -MD -MT CMakeFiles/cmTC_2e99a.dir/CMakeCCompilerABI.c.o -MF CMakeFiles/cmTC_2e99a.dir/CMakeCCompilerABI.c.o.d -o CMakeFiles/cmTC_2e99a.dir/CMakeCCompilerABI.c.o -c /opt/homebrew/Cellar/cmake/3.25.1/share/cmake/Modules/CMakeCCompilerABI.c
Apple clang version 14.0.0 (clang-1400.0.29.202)
Target: unknown-apple-darwin22.2.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
clang: error: invalid arch name '-arch “x86_64”'
clang: warning: -Wl,-v: 'linker' input unused [-Wunused-command-line-argument]
gmake[1]: *** [CMakeFiles/cmTC_2e99a.dir/build.make:79: CMakeFiles/cmTC_2e99a.dir/CMakeCCompilerABI.c.o] Error 1
gmake[1]: Leaving directory '/Users/user/acados/build/CMakeFiles/CMakeScratch/TryCompile-ywRMou'
gmake: *** [Makefile:127: cmTC_2e99a/fast] Error 2

Determining if the C compiler works failed with the following output:
Change Dir: /Users/user/acados/build/CMakeFiles/CMakeScratch/TryCompile-rLynz3

Run Build Command(s):/opt/homebrew/bin/gmake -f Makefile cmTC_e227e/fast && /opt/homebrew/bin/gmake  -f CMakeFiles/cmTC_e227e.dir/build.make CMakeFiles/cmTC_e227e.dir/build
gmake[1]: Entering directory '/Users/user/acados/build/CMakeFiles/CMakeScratch/TryCompile-rLynz3'
Building C object CMakeFiles/cmTC_e227e.dir/testCCompiler.c.o
/Library/Developer/CommandLineTools/usr/bin/cc   -arch “x86_64” -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.1.sdk -MD -MT CMakeFiles/cmTC_e227e.dir/testCCompiler.c.o -MF CMakeFiles/cmTC_e227e.dir/testCCompiler.c.o.d -o CMakeFiles/cmTC_e227e.dir/testCCompiler.c.o -c /Users/user/acados/build/CMakeFiles/CMakeScratch/TryCompile-rLynz3/testCCompiler.c
clang: error: invalid arch name '-arch “x86_64”'
gmake[1]: *** [CMakeFiles/cmTC_e227e.dir/build.make:79: CMakeFiles/cmTC_e227e.dir/testCCompiler.c.o] Error 1
gmake[1]: Leaving directory '/Users/user/acados/build/CMakeFiles/CMakeScratch/TryCompile-rLynz3'
gmake: *** [Makefile:127: cmTC_e227e/fast] Error 2
```

FreyJo, 2022-12-21:
Did you try using the CMake arguments from the thread?
`-DCMAKE_OSX_ARCHITECTURES="x86_64"`

Mathiasschuler96, 2022-12-21:
I tried to install acados as followed:
```
mkdir -p build
cd build
cmake -DACADOS_WITH_QPOASES=ON 
cmake -DCMAKE_OSX_ARCHITECTURES="x86_64" 
make install -j4
```
I got the following error:
```
CMake Warning:
  No source or binary directory provided.  Both will be assumed to be the
  same as the current working directory, but note that this warning will
  become a fatal error in future CMake releases.

CMake Error: The source directory "/Users/user/acados/build" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
```
Am I doing something wrong?

Mathiasschuler96, 2023-02-05:
I was still not able to run the example... I tried almost every possible way to install acados. Any advice?

gy2256, 2024-02-29:
I got the python example working with Apple Silicon M3 Max after making some changes to the source code to get it compiled. However, I forgot what I changed to get it working. 
[image]