acados/acados #905: Mismatching dimension errors with thin wrapper class around AcadosOcpSolver

spenrich, 2023-01-05:
I'm on Linux (Ubuntu 22.04). I've installed Acados via cmake with the command:
```Shell
cmake -DACADOS_WITH_OSQP=ON -DACADOS_WITH_QPOASES=ON -DACADOS_WITH_OPENMP=ON ..
```
I've added the following to my .zshrc file:
```Shell
export ACADOS_SOURCE_DIR="$HOME/src/acados"
export LD_LIBRARY_PATH="$ACADOS_SOURCE_DIR/lib:$LD_LIBRARY_PATH"
```
I've successfully installed the Python interface and tested a number of the provided examples. 

I've put together the following MWE to demonstrate an issue I'm having. It is based on the cart-pole example [here](https://github.com/acados/acados/blob/master/examples/acados_python/getting_started/minimal_example_ocp.py).
```Python
import os
import shutil
from acados_template import (AcadosModel, AcadosOcp, AcadosOcpSolver,
                             ocp_get_default_cmake_builder)
import casadi as cs
import numpy as np

class OCPCartPole:
    """Wrapper around an OCP for the cart-pole."""

    def __init__(self, num_stages: int, T: float,
                 code_export_directory: str = 'c_generated_code',
                 json_file: str = 'acados_ocp.json',
                 use_cmake: bool = False):
        """Initialize; see `AcadosCartPole`."""
        self.dims = (4, 1)
        self.num_stages = num_stages
        self.T = T
        self.code_export_dir = code_export_directory
        self.json_file = json_file
        self.use_cmake = use_cmake
        self.solver = self.construct_solver()

    def construct_solver(self):
        """Construct an Acados solver."""
        # Constants
        mc = 1.                                 # mass of the cart [kg]
        mp = 0.1                                # mass of the pendulum [kg]
        g = 9.81                                # gravity constant [m/s^2]
        L = 0.8                                 # length of the rod [m]
        F_max = 80.                             # control magnitude bound
        Q = np.diag([1e3, 1e3, 1e-2, 1e-2])     # state weight matrix
        R = np.diag([1e-2])                     # control weight matrix
        n, m = self.dims                        # state and control dimensions
        T = self.T                              # time horizon
        N = self.num_stages                     # discrete prediction horizon

        # Dynamics model
        x = cs.SX.sym('x', n)  # (p, θ, dp, dθ)
        u = cs.SX.sym('x', m)
        θ, dp, dθ = x[1], x[2], x[3]
        cosθ, sinθ = cs.cos(θ), cs.sin(θ)
        h = mc + mp*(1. - cosθ**2)
        f = cs.vertcat(dp,
                       dθ,
                       (-mp*L*sinθ*(dθ**2) + mp*g*cosθ*sinθ + u)/h,
                       (-mp*cosθ*sinθ*(dθ**2) + (mc+mp)*g*sinθ/L + u*cosθ/L)/h)
        model = AcadosModel()
        model.name = 'pendulum_ode'
        model.x = x
        model.u = u
        model.f_expl_expr = f

        # OCP
        ocp = AcadosOcp()
        ocp.code_export_directory = code_export_directory
        ocp.model = model
[... 165 lines cut ...]
[ 78%] Building C object CMakeFiles/ocp_pendulum_ode.dir/pendulum_ode_cost/pendulum_ode_cost_ext_cost_fun_jac_hess.c.o
[ 78%] Building C object CMakeFiles/ocp_pendulum_ode.dir/pendulum_ode_cost/pendulum_ode_cost_ext_cost_0_fun_jac.c.o
[ 92%] Building C object CMakeFiles/model_pendulum_ode.dir/pendulum_ode_model/pendulum_ode_expl_ode_hess.c.o
[ 92%] Building C object CMakeFiles/ocp_pendulum_ode.dir/acados_solver_pendulum_ode.c.o
[ 92%] Built target model_pendulum_ode
[ 92%] Built target ocp_pendulum_ode
[100%] Linking C shared library libacados_ocp_solver_pendulum_ode.so
[100%] Built target acados_ocp_solver_pendulum_ode
call("cmake --install "/home/spenrich/src/mwe_acados/c_generated_code/build"")
-- Install configuration: "Release"
-- Installing: /home/spenrich/src/mwe_acados/c_generated_code/libacados_ocp_solver_pendulum_ode.so
-- Set runtime path of "/home/spenrich/src/mwe_acados/c_generated_code/libacados_ocp_solver_pendulum_ode.so" to ""
acados was compiled with OpenMP.
Traceback (most recent call last):
  File "/home/spenrich/src/mwe_acados/cartpole.py", line 147, in <module>
    solver.solve(x0, xT)
  File "/home/spenrich/src/mwe_acados/cartpole.py", line 116, in solve
    solver.constraints_set(N, 'lbx', xT)
  File "/home/spenrich/src/acados/interfaces/acados_template/acados_template/acados_ocp_solver.py", line 1782, in constraints_set
    raise Exception(f'AcadosOcpSolver.constraints_set(): mismatching dimension' +
Exception: AcadosOcpSolver.constraints_set(): mismatching dimension for field "lbx" at stage c_int(20) with dimension (0, 0) (you have (4, 0))
```
From the output, we can see that the first run proceeds successfully, but the second run causes a strange dimension error. 

I can't see why `lbx` would ever have a dimension of 0, especially since the first run is successful. Any idea why this might be happening?

FreyJo, 2023-01-06:
Hi,
I just checked the example.
The issue is that the `solver` object is not fully destroyed in your example.
It works fine with this modification.
```
    # Solve
...
    del solver

    # Clear files and solve again with a different `N`
...
```

Cheers!

spenrich, 2023-01-06:
@FreyJo thank you for the tip and your time! 

While this issue has been closed, I hope you don't mind if I continue to report on this to help solve my use-case (and potentially anyone else's in case they stumble upon this). I'd like to make this thin wrapper very straightforward to use (e.g., by a student in an introductory course on optimal control), so forcing the user to remember to explicitly delete a Python object (which is not normally required) is not ideal.

Based on the warning message in the `__del__` method of `AcadosOcpSolver` [here](https://github.com/acados/acados/blob/59a98cbefdee4f7017c9e4e4993fdfc76442b14c/interfaces/acados_template/acados_template/acados_ocp_solver.py#L1953-L1968), I re-wrote my example to allow changing the model name.
```Python
import os
import shutil
from acados_template import (AcadosModel, AcadosOcp, AcadosOcpSolver,
                             ocp_get_default_cmake_builder)
import casadi as cs
import numpy as np

class OCPCartPole:
    """Wrapper around an OCP for the cart-pole."""

    def __init__(self, num_stages: int, T: float,
                 code_export_directory: str = 'c_generated_code',
                 json_file: str = 'acados_ocp.json',
                 use_cmake: bool = False,
                 model_name: str = 'cartpole_ode'):
        """Initialize; see `OCPCartPole`."""
        self.dims = (4, 1)
        self.num_stages = num_stages
        self.T = T
        self.code_export_dir = code_export_directory
        self.json_file = json_file
        self.use_cmake = use_cmake
        self.model_name = model_name
        self.solver = self.construct_solver()

    def construct_solver(self):
        """Construct an Acados solver."""
        # Constants
        mc = 1.                                 # mass of the cart [kg]
        mp = 0.1                                # mass of the pendulum [kg]
        g = 9.81                                # gravity constant [m/s^2]
        L = 0.8                                 # length of the rod [m]
        F_max = 80.                             # control magnitude bound
        Q = np.diag([1e3, 1e3, 1e-2, 1e-2])     # state weight matrix
        R = np.diag([1e-2])                     # control weight matrix
        n, m = self.dims                        # state and control dimensions
        T = self.T                              # time horizon
        N = self.num_stages                     # discrete prediction horizon

        # Dynamics model
        x = cs.SX.sym('x', n)  # (p, θ, dp, dθ)
        u = cs.SX.sym('x', m)
        θ, dp, dθ = x[1], x[2], x[3]
        cosθ, sinθ = cs.cos(θ), cs.sin(θ)
        h = mc + mp*(1. - cosθ**2)
        f = cs.vertcat(dp,
                       dθ,
                       (-mp*L*sinθ*(dθ**2) + mp*g*cosθ*sinθ + u)/h,
                       (-mp*cosθ*sinθ*(dθ**2) + (mc+mp)*g*sinθ/L + u*cosθ/L)/h)
        model = AcadosModel()
        model.name = self.model_name
        model.x = x
        model.u = u
        model.f_expl_expr = f

        # OCP
        ocp = AcadosOcp()
        ocp.code_export_directory = self.code_export_dir
        ocp.model = model
        ocp.dims.N = N
        ocp.solver_options.tf = T

        # Cost function
        ocp.cost.cost_type = 'EXTERNAL'
[... 182 lines cut ...]
acados was compiled without OpenMP.

********* CALLING DESTRUCTOR! *********

Breakpoint here (check if destructor of first `solver` is called).
Traceback (most recent call last):
  File "/home/spenrich/src/mwe_acados/cartpole.py", line 159, in <module>
    solver.solve(x0, xT)
  File "/home/spenrich/src/mwe_acados/cartpole.py", line 118, in solve
    solver.constraints_set(N, 'lbx', xT)
  File "/home/spenrich/src/acados/interfaces/acados_template/acados_template/acados_ocp_solver.py", line 1782, in constraints_set
    raise Exception(f'AcadosOcpSolver.constraints_set(): mismatching dimension' +
Exception: AcadosOcpSolver.constraints_set(): mismatching dimension for field "lbx" at stage c_int(30) with dimension (0, 0) (you have (4, 0))

********* CALLING DESTRUCTOR! *********
```

I've added the destructor message to see when the first `solver` is destroyed. Interestingly, it seems to be destroyed _after_ the build/linking process for the second `solver`. So I think the error I'm getting when the model name is the same is caused precisely by the issue described in the `__del__` method of `AcadosOcpSolver`. 

Overall, I think what happens is:
1. The second instance of `OCPCartPole` is initialized and re-uses the old Acados library (since it has not been closed yet).
2. The name `solver` is given to the second instance of `OCPCartPole`, and so the destructor of the old `solver` is called. This calls the `__del__` method of `AcadosOcpSolver`, which somehow causes there to be "zero dimensions" (e.g., see the error `Exception: AcadosOcpSolver.constraints_set(): mismatching dimension for field "lbx" at stage c_int(30) with dimension (0, 0) (you have (4, 0))`).
3. Trying to solve the OCP problem again raises this error.

Does my understanding of this seem correct / incorrect? Do you have any suggestions for how to circumvent this issue without requiring the user to explicitly delete objects? I don't mind baking careful handling of library objects, object deletion, etc. into the source code of `OCPCartPole`, but usage of `OCPCartPole` should be as mundane as possible.

FreyJo, 2023-01-09:
I reopened, because I would also be really happy to have a fix for this.

> The name `solver` is given to the second instance of `OCPCartPole`, and so the destructor of the old `solver` is called. This calls the `__del__` method of `AcadosOcpSolver`, which somehow causes [...]

I think it is a bit different:
The destructor is not the problem, but the setter `solver.constraints_set(N, 'ubx', xT)`, uses the old solver lib, because it was not unloaded yet.

The underlying issue is, that first the second `solver` is created, then the first `solver` is deleted.
I don't know why Python does this and if one can change it..
Removing the generated code directory doesn't really do anything, the library is still loaded anyway

Here is my modified code:

```
import os
import shutil
from acados_template import (AcadosModel, AcadosOcp, AcadosOcpSolver,
                             ocp_get_default_cmake_builder)
import casadi as cs
import numpy as np

class OCPCartPole:
    """Wrapper around an OCP for the cart-pole."""

    def __init__(self, num_stages: int, T: float,
                 code_export_directory: str = 'c_generated_code',
                 json_file: str = 'acados_ocp.json',
                 use_cmake: bool = False,
                 model_name: str = 'cartpole_ode'):
        """Initialize; see `OCPCartPole`."""
        self.dims = (4, 1)
        self.num_stages = num_stages
        self.T = T
        self.code_export_dir = code_export_directory
        self.json_file = json_file
        self.use_cmake = use_cmake
        self.model_name = model_name
        self.solver = self.construct_solver()
        print("end __init__ OCPCartPole")

    def construct_solver(self):
        """Construct an Acados solver."""
        # Constants
        mc = 1.                                 # mass of the cart [kg]
        mp = 0.1                                # mass of the pendulum [kg]
        g = 9.81                                # gravity constant [m/s^2]
        L = 0.8                                 # length of the rod [m]
        F_max = 80.                             # control magnitude bound
        Q = np.diag([1e3, 1e3, 1e-2, 1e-2])     # state weight matrix
        R = np.diag([1e-2])                     # control weight matrix
        n, m = self.dims                        # state and control dimensions
        T = self.T                              # time horizon
        N = self.num_stages                     # discrete prediction horizon

        # Dynamics model
        x = cs.SX.sym('x', n)  # (p, θ, dp, dθ)
        u = cs.SX.sym('x', m)
        θ, dp, dθ = x[1], x[2], x[3]
        cosθ, sinθ = cs.cos(θ), cs.sin(θ)
        h = mc + mp*(1. - cosθ**2)
        f = cs.vertcat(dp,
                       dθ,
                       (-mp*L*sinθ*(dθ**2) + mp*g*cosθ*sinθ + u)/h,
                       (-mp*cosθ*sinθ*(dθ**2) + (mc+mp)*g*sinθ/L + u*cosθ/L)/h)
        model = AcadosModel()
        model.name = self.model_name
        model.x = x
        model.u = u
        model.f_expl_expr = f

[... 76 lines cut ...]

    # Solve
    model_name = 'cartpole_ode'
    N = 40
    if os.path.isdir(code_export_directory):
        shutil.rmtree(code_export_directory)
    solver = OCPCartPole(N, T, code_export_directory, json_file, use_cmake,
                         model_name)
    shutil.rmtree(code_export_directory)
    solver.solve(x0, xT)

    # del solver
    # Clear files and solve again with a different `N`

    # code_export_directory += '1'
    json_file = os.path.join(code_export_directory, 'acados_ocp.json')
    model_name = 'cartpole_ode_2'
    N = 30
    solver = OCPCartPole(N, T, code_export_directory, json_file, use_cmake,
                         model_name)

    print('Breakpoint here (check if destructor of first `solver` is called).')

    solver.solve(x0, xT)
```