acados/acados #912: Bug in eval_param_sens ?

andretag, 2023-02-09:
I am wondering if there is a bug in the output of `eval_param_sens`, because when I evaluate the sensitivity of the solution found at the origin ([0., 0,], considering an MPC for a double integrator), I obtain a value that does not match the slope of the solution found solving the full mpc problem in the neighborhood of that point. 

A plot highlighting the issue is attached, showing the optimal u from mpc ("full mpc solution") as a function of one of the two components of the state (`x_1`, while `x_2=0`) and the slope obtained from the sensitivity at the origin of the state space.

[image]

The issue does not appear at other points: 
[image]
 
Attached you can find the Jupiter notebook used to generate the plots 
[acados_sensitivity_test.ipynb.zip](https://github.com/acados/acados/files/10698959/acados_sensitivity_test.ipynb.zip)

Happy to open a PR to fix, if pointed in the right direction.

giaf, 2023-03-01:
The sensitivity is based on the latest internal linearization of the KKT conditions performed in HPIPM.
If the last QP did not converge to high accuracy, the sensitivity may be inaccurate.

In this case, the system to be controlled is a double integrator with reference [0,0].
When the MPC problem is solved with x0=[0,0], what is it happening? Are some QPs still solved? To what accuracy?

FreyJo, 2023-03-02:
I had a chat with @giaf and we found what the issue is. I'll try to summarize:

The problem in your example is, that the initial guess is already the solution of the OCP in the zero position.
Thus, in SQP the residuals fulfill the tolerance before a QP is solved.
However, `eval_param_sens` function assumes:
1. that the problem is solved for the current data and
2. that HPIPM has a factorized KKT system at the solution in memory.
In your example, the latter is not the case.
You can verify this, by just removing the `load_iterate` in your example, in this case you don't initialize at the solution and the sensitivities are correct.
Note, that this is a very special case in practice.

A possible fix would be to check if the QP solver got called in the previous SQP call and if not, call HPIPM once before the other code to compute the sensitivities is executed.

andretag, 2023-03-10:
@FreyJo and @giaf, thanks a lot for your reply. 

Is there a way to make the call to HPIPM from the pyhton interface? 

Do I also understand correctly that a possible way to check if the previous QP solver got called in the previous SQP call is by checking that last entry of the vector in  'qp_iter' is equal to 1 ?

Thanks again!

FreyJo, 2025-02-10:
Sorry for the long inactivity on this topic.

> Is there a way to make the call to HPIPM from the pyhton interface?

No, this is not possible right now.
There are two issues:
1) the SQP solver exits before calling the QP solver
2) the QP solver exits before factorizing the matrix
Both can happen if the solver is initialized too closed to the solution, depending on the tolerances.

The functionality `setup_qp_matrices_and_factorize()` added in https://github.com/acados/acados/pull/1432 can be used to mitigate 1), by calling it if the solver exits without performing iterations.

In order to mitigate 2), the option  `update_fact_exit` was added in HPIPM. This option is on by default, so 2) should not be a problem anymore.

I hope you are still interested in using acados for this. Your feedback is highly appreciated!

closed 2025-02-10 by FreyJo