acados/acados #936: question about nmpc

JacksinU, 2023-06-13:
Hello everyone,

A paper says that admm and ipopt are covered in this powerful toolbox. I heard that admm is fast for nmpc. ADMM doesn't seem to be included according to documentation. May I know whether admm is supported or not? 

There are some examples using nmpc like race cars in python, but most of them assume that the control horizon is the same as the prediction horizon. Is there any example for nmpc where control horizon is shorter than prediction horizon?

FreyJo, 2023-06-14:
Hi,
can you please mention the paper you are referring to?
`acados` does not offer an interface.
The only interface to a first-order QP solver is OSQP.
There is no ADMM solver interfaced.

The other questions are answered already:
https://discourse.acados.org/t/control-horizon-and-prediction-horizon/442

ACADO and acados share no code at all:
https://docs.acados.org/faq/index.html#faq

More documentation can be found here:
https://docs.acados.org/

JacksinU, 2023-06-14:
> Hi, can you please mention the paper you are referring to? `acados` does not offer an interface. The only interface to a first-order QP solver is OSQP. There is no ADMM solver interfaced.
>
> The other [...]

Hi FreyJo,

Thanks for your link. Acado and ADMM are mentioned in Table 4 in paper Model predictive path tracking control for automated road vehicles: A review.

JacksinU, 2023-06-25:
Hi everyone,

I am new to this powful toolbox and still learning the basics. If integrator_type in AcadosOcpOptions is defined as 'ERK', is it RK4？Is discrete Dynamics required in f_impl_expr and f_expl_expr when integrator_type is 'DISCRETE'?

closed 2023-06-26 by JacksinU