acados/acados #956: Unable to import acados

rakeshv24, 2023-08-14:
I followed the installation instructions mentioned in the documentation but I'm still not able to use the package. Current system specifications:
- Ubuntu 20.04
- Dell Alienware m15 r4
- Python 3.8

I'm able to run the example python interface file but not able to import acados or any modules such as ocp, ocp_nlp, ocp_qp.

Any help would be appreciated.
Thank you!

FreyJo, 2023-08-14:
Did you compile the libraries successfully?

> not able to import acados or any modules such as ocp, ocp_nlp, ocp_qp

These are not things you can import from the acados python interface.

rakeshv24, 2023-08-14:
Yes, the CMake build process was successful without any errors or warnings. Is there more to the compilation process?

FreyJo, 2023-08-14:
yes, you also have to run make. Please check the instructions closely again.

rakeshv24, 2023-08-14:
It was unclear from the instructions that both cmake and make have to be run. The following was mentioned in the documentation: 

"Both a CMake and a Makefile based build system is supported at the moment. Please **choose one** and proceed with the corresponding paragraph."

I still went ahead and executed instructions for cmake and make as you suggested. But the problem still exists. Unable to import the following: 
`from acados import ocp, ocp_nlp, ocp_qp`

Additionally, `minimal_example_ocp.py` doesn't work now. It throws the following error: `OSError: /home/darth/acados/lib/libacados.so: undefined symbol: osqp_update_P_A
`

rakeshv24, 2023-08-24:
Any help would be appreciated! :)

FreyJo, 2023-08-24:
Why do you think this should work?

> `from acados import ocp` => `ImportError: cannot import name 'ocp' from 'acados' (unknown location)`
`from acados import ocp_qp_solver` => `ImportError: cannot import name 'ocp_qp_solver' from 'acados' (unknown location)`

The acados python module is called `acados_template` it does not expose `ocp` or `ocp_ocp_qp_solver`.

About the OSQP error: I am not sure. Can you try to compile `acados` with OSQP?
Or do you have the lib compiled already. Check what is in `acados_root_dir/lib`.

closed 2023-08-29 by sandmaennchen