acados/acados #957: Unable to run the  sqp_rti framework in c code with acados model generated by Python

WantStudyMagicInlaoshan, 2023-08-15:
Hi,
I followed the (acados_template/c_templates_tera/main.in.c)'s instructions to build my c code, and I have encountered an issue iwhile using the acados_model_solver exported by python.

It reports "segmentation fault" when run the SQP solving code "status = {model_name}_acados_solve(acados_ocp_capsule);".

Can you provide some solutions of the above issue?

Thank you!

FreyJo, 2023-08-15:
Hi,

which example did you run and get this issue in the corresponding generated `main.c` file.
I suggest running it with debug flags and valgrind to find a potential memory issue.
Before make sure though that you installed `acados` fresh after pulling the latest version.

Cheers!

WantStudyMagicInlaoshan, 2023-08-15:
Hi,

Firstly, I constructed the OCP problem using casadi and acados in Python and generated libacados_ OCP_ Solver.so;

Aiming to develop this SQP_RTI solver  in c environment, I developed my own main.cpp (not -git pull) based on the template in interfaces/acados_template/acados_template/c_templates_tera/main.in.c .

Previously, all the required libs have been confirmed to be connected (Cmakelist compilation passed)

i have nothing to do about this, can you provide some solutions？

closed 2023-08-16 by LilHu7