acados/acados #966: Cython interface doesn't build on macOS

tudoroancea, 2023-09-01:
Hi :wave: 

I have tried for quite some time to use the Cython interface on macOS (13.5.1 on apple silicon M2) but did not have any success yet, and thought I would share my journey in case anyone shared my struggles. Concretely, when trying to build even the simple examples such as `examples/acados_python/pendulum_on_cart/cython_example_closed_loop.py`, I encounter the following type of error:

<details>
  <summary>Error 1</summary>

```
arm64-apple-darwin20.0.0-clang -fPIC -std=c99   -O2 -D_FORTIFY_SOURCE=2 -isystem /Users/tudoroancea/miniforge3/envs/acados/include -I/Users/tudoroancea/Developer/acados/include -I/Users/tudoroancea/Developer/acados/include/acados -I/Users/tudoroancea/Developer/acados/include/blasfeo/include -I/Users/tudoroancea/Developer/acados/include/hpipm/include  -c -o acados_solver_pendulum.o acados_solver_pendulum.c
arm64-apple-darwin20.0.0-clang -fPIC -std=c99   -O2 -D_FORTIFY_SOURCE=2 -isystem /Users/tudoroancea/miniforge3/envs/acados/include -I/Users/tudoroancea/Developer/acados/include -I/Users/tudoroancea/Developer/acados/include/acados -I/Users/tudoroancea/Developer/acados/include/blasfeo/include -I/Users/tudoroancea/Developer/acados/include/hpipm/include  -c -o pendulum_model/pendulum_expl_ode_fun.o pendulum_model/pendulum_expl_ode_fun.c
arm64-apple-darwin20.0.0-clang -fPIC -std=c99   -O2 -D_FORTIFY_SOURCE=2 -isystem /Users/tudoroancea/miniforge3/envs/acados/include -I/Users/tudoroancea/Developer/acados/include -I/Users/tudoroancea/Developer/acados/include/acados -I/Users/tudoroancea/Developer/acados/include/blasfeo/include -I/Users/tudoroancea/Developer/acados/include/hpipm/include  -c -o pendulum_model/pendulum_expl_vde_forw.o pendulum_model/pendulum_expl_vde_forw.c
arm64-apple-darwin20.0.0-clang -fPIC -std=c99   -O2 -D_FORTIFY_SOURCE=2 -isystem /Users/tudoroancea/miniforge3/envs/acados/include -I/Users/tudoroancea/Developer/acados/include -I/Users/tudoroancea/Developer/acados/include/acados -I/Users/tudoroancea/Developer/acados/include/blasfeo/include -I/Users/tudoroancea/Developer/acados/include/hpipm/include  -c -o pendulum_model/pendulum_expl_vde_adj.o pendulum_model/pendulum_expl_vde_adj.c
arm64-apple-darwin20.0.0-clang -shared acados_solver_pendulum.o pendulum_model/pendulum_expl_ode_fun.o pendulum_model/pendulum_expl_vde_forw.o pendulum_model/pendulum_expl_vde_adj.o -o libacados_ocp_solver_pendulum.dylib -Wl,-pie -Wl,-headerpad_max_install_names -Wl,-dead_strip_dylibs -Wl,-rpath,/Users/tudoroancea/miniforge3/envs/acados/lib -L/Users/tudoroancea/miniforge3/envs/acados/lib -L/Users/tudoroancea/Developer/acados/lib -lacados -lhpipm -lblasfeo -lm  \
	-L -l
cython \
	-o acados_ocp_solver_pyx.c \
	-I /Users/tudoroancea/Developer/acados/include/../interfaces/acados_template/acados_template \
	/Users/tudoroancea/Developer/acados/include/../interfaces/acados_template/acados_template/acados_ocp_solver_pyx.pyx \
	-I /Users/tudoroancea/Developer/acados/examples/acados_python/pendulum_on_cart/c_generated_code \

arm64-apple-darwin20.0.0-clang  -c -O2 \
	-fPIC \
	-o acados_ocp_solver_pyx.o \
	-I /Users/tudoroancea/Developer/acados/include/blasfeo/include/ \
	-I /Users/tudoroancea/Developer/acados/include/hpipm/include/ \
	-I /Users/tudoroancea/Developer/acados/include \
	-I /Users/tudoroancea/miniforge3/envs/acados/lib/python3.10/site-packages/numpy/core/include \
	-I /Users/tudoroancea/miniforge3/envs/acados/include/python3.10 \
	acados_ocp_solver_pyx.c \

arm64-apple-darwin20.0.0-clang  -shared \
	-o acados_ocp_solver_pyx.dylib \
	-Wl,-rpath=/Users/tudoroancea/Developer/acados/lib \
	acados_ocp_solver_pyx.o \
	/Users/tudoroancea/Developer/acados/examples/acados_python/pendulum_on_cart/c_generated_code/libacados_ocp_solver_pendulum.dylib \
	-Wl,-pie -Wl,-headerpad_max_install_names -Wl,-dead_strip_dylibs -Wl,-rpath,/Users/tudoroancea/miniforge3/envs/acados/lib -L/Users/tudoroancea/miniforge3/envs/acados/lib -L/Users/tudoroancea/Developer/acados/lib -lacados -lhpipm -lblasfeo -lm 
Trying to load 'c_generated_code.acados_ocp_solver_pyx' while in directory /Users/tudoroancea/Developer/acados/examples/acados_python/pendulum_on_cart
```

</details>

Basically, the linking phase of the `pyx` dylib creation fails, which makes its loading also fail. As I use `clang` and not `gcc`, I have followed this [stackoverflow post](https://stackoverflow.com/questions/6562403/i-dont-understand-wl-rpath-wl) and tried modifying the template `Makefile.in` in the following way:

<details>
<summary>Error 2</summary>
```
diff --git a/interfaces/acados_template/acados_template/c_templates_tera/Makefile.in b/interfaces/acados_template/acados_template/c_templates_tera/Makefile.in
index fbefc08e..3d3510d5 100644
--- a/interfaces/acados_template/acados_template/c_templates_tera/Makefile.in
+++ b/interfaces/acados_template/acados_template/c_templates_tera/Makefile.in
@@ -386,7 +386,7 @@ ocp_cython_o: ocp_cython_c
 ocp_cython: ocp_cython_o
        $(CC) $(ACADOS_FLAGS) -shared \
        -o acados_ocp_solver_pyx{{ shared_lib_ext }} \
-       -Wl,-rpath=$(LIB_PATH) \
+       -Wl,-rpath,$(LIB_PATH) \
        acados_ocp_solver_pyx.o \
        $(abspath .)/libacados_ocp_solver_{{ model.name }}{{ shared_lib_ext }} \
        $(LDFLAGS) $(LDLIBS)
@@ -414,7 +414,7 @@ sim_cython_o: sim_cython_c
 sim_cython: sim_cython_o
        $(CC) $(ACADOS_FLAGS) -shared \
        -o acados_sim_solver_pyx{{ shared_lib_ext }} \
-       -Wl,-rpath=$(LIB_PATH) \
+       -Wl,-rpath,$(LIB_PATH) \
        acados_sim_solver_pyx.o \
        $(abspath .)/libacados_sim_solver_{{ model.name }}{{ shared_lib_ext }} \
        $(LDFLAGS) $(LDLIBS)
```
It effectively solved the first linking error, but another one appeared:
[... 820 lines cut ...]
      ...
  "__Py_TrueStruct", referenced from:
      ___pyx_pymod_exec_acados_ocp_solver_pyx in acados_ocp_solver_pyx.o
      ___Pyx_AddTraceback in acados_ocp_solver_pyx.o
      ___Pyx_ImportDottedModule in acados_ocp_solver_pyx.o
      ___Pyx_CyFunction_get_is_coroutine in acados_ocp_solver_pyx.o
      ___pyx_array_get_memview in acados_ocp_solver_pyx.o
      ___pyx_memoryview_is_slice in acados_ocp_solver_pyx.o
      ___pyx_pw_15acados_template_21acados_ocp_solver_pyx_21AcadosOcpSolverCython_17eval_param_sens in acados_ocp_solver_pyx.o
      ...
ld: symbol(s) not found for architecture arm64
clang-15: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [ocp_cython] Error 1
```
</details>

From this [other post](https://stackoverflow.com/questions/55528299/cython-link-errors-for-osx), I think it might be due to the mac framework files necessary for python. I tried adding as suggested the output of `python3-config --ldflags` and `python3-config --cflags` to the compiler command (again directly in `Makefile.in`) but the error remains.

Any ideas?

Thank you in advance for your help :pray: 

Best,

Ted

aghezz1, 2023-09-04:
Hi Ted

I have macOS 12.3.1 on Intel. I don't think the CPU makes a different here.
I have the same issue of you and I cannot completely fix it but I can at least build the example and generate the `.dylib` file.

1. I use `gcc-10` as default compiler to avoid issues when building "Linux-based" projects, just added `export CC=/usr/local/bin/gcc-10` in my zsh profile
2. To create the `.dylib` file I simply commented line [L389](https://github.com/acados/acados/blob/9622cfdebdb06a91a4a65bf4fe42adf2a8f16152/interfaces/acados_template/acados_template/c_templates_tera/Makefile.in#L389). Otherwise it raises the following error: `ld: unknown option: -rpath=/Users/aghezzi/acados/lib
collect2: error: ld returned 1 exit status
make: *** [ocp_cython] Error 1`
3. But now I have another problem, when I try to import the compiled library as python module I have a `ModuleNotFoundError: No module named 'c_generated_code.acados_ocp_solver_pyx'`

Up to now I have not understood this last issue, I will post here if I have updates.

Hope my small patch can be useful to you, let me know if you make progress 😄 

Best,
Andrea

gy2256, 2023-09-07:
I built it successfully on MacBook Pro with M1 Pro and Mac OS Ventura 13.5.1. But I had to add a bunch of `#include <stdlib.h>` in multiple source files to address the malloc errors.

aghezz1, 2023-09-11:
Hi Guang, 
thanks for your comment, could you share your edits via a PR?
Unfortunately, I think my current error is due to python `import module` routines and not to malloc errors.
Best,
Andrea

gy2256, 2023-10-03:
Hello Andrea, 

Sorry about the late reply. I was working on several projects during the past few weeks. 

Yeah, I can create a PR over the weekend. 

Best,
Guang