acados/acados #972: macOS (M1) installation issue with MAKE

jy-cds, 2023-10-04:
I followed the installation instruction [here](https://docs.acados.org/installation/index.html), added "ARMV8A_APPLE_M1" to the BLASFEO_TARGET and changed ACADOS_WITH_QPOASES = 1. However, when I try to run "make shared_library" I encountered the following error:

echo "{" > ./lib/link_libs.json
/bin/sh: ./lib/link_libs.json: No such file or directory
make: *** [link_libs_json] Error 1

Any help is much appreciated. Thanks!

Jing

FreyJo, 2023-10-04:
Hi Jing,

Maybe you just need to create the `lib` directory.

In general, I think the acados CMake build process is better tested and some people used it successfully to compile on M1 Mac.
Is there any reason you want to stick to Make?

Best,
Jonathan

jy-cds, 2023-10-04:
Hi Jonathan,

Thank you for the reply. I will try that. The reason I went with MAKE instead of CMAKE is because (1) in the BLASFEO repo, the authors write that apple M1 is not currently supported by CMAKE, and (2) when I did try CMAKE following the instructions, it say 
`CMake Error: The source directory "/Users/jing/acados/build" does not appear to contain CMakeLists.txt.`
even though I passed the folder arguments already, i.e. 
`cmake -DACADOS_WITH_QPOASES=ON -DACADOS_INSTALL_DIR=/Users/jing/acados`.
I also tried CMAKE directly in the "acados" root folder, but I ran into other errors when I was trying out the examples. 

Best,
Jing

jy-cds, 2023-10-04:
MAKE updates: after creating a `lib` folder in the `acados` root folder, I ran `make shared_library` in the command line and get the following error output:

> echo "{" > ./lib/link_libs.json
echo "\t\"openmp\": \"\"," >> ./lib/link_libs.json
echo "\t\"qpoases\": \"-lqpOASES_e\"," >> ./lib/link_libs.json
echo "\t\"qpdunes\": \"\"," >> ./lib/link_libs.json
echo "\t\"osqp\": \"\"," >> ./lib/link_libs.json
echo "\t\"hpmpc\": \"\"," >> ./lib/link_libs.json
echo "\t\"ooqp\": \"\"" >> ./lib/link_libs.json
echo "}" >> ./lib/link_libs.json
( cd /Users/jing/acados/external/blasfeo; /Applications/Xcode.app/Contents/Developer/usr/bin/make shared_library CC=gcc LA=HIGH_PERFORMANCE TARGET=X64_INTEL_HASWELL MF=PANELMAJ BLAS_API=0 )
Parsing Makefile.rule
touch ./include/blasfeo_target.h
echo "#ifndef TARGET_X64_INTEL_HASWELL" >  ./include/blasfeo_target.h
echo "#define TARGET_X64_INTEL_HASWELL" >> ./include/blasfeo_target.h
echo "#endif"                           >> ./include/blasfeo_target.h
echo "#ifndef TARGET_NEED_FEATURE_AVX2" >> ./include/blasfeo_target.h
echo "#define TARGET_NEED_FEATURE_AVX2" >> ./include/blasfeo_target.h
echo "#endif"                           >> ./include/blasfeo_target.h
echo "#ifndef TARGET_NEED_FEATURE_FMA"  >> ./include/blasfeo_target.h
echo "#define TARGET_NEED_FEATURE_FMA"  >> ./include/blasfeo_target.h
echo "#endif"                           >> ./include/blasfeo_target.h
echo "#ifndef LA_HIGH_PERFORMANCE" >> ./include/blasfeo_target.h
echo "#define LA_HIGH_PERFORMANCE" >> ./include/blasfeo_target.h
echo "#endif"                      >> ./include/blasfeo_target.h
echo "#ifndef MF_PANELMAJ" >> ./include/blasfeo_target.h
echo "#define MF_PANELMAJ" >> ./include/blasfeo_target.h
echo "#endif"              >> ./include/blasfeo_target.h
echo "#ifndef EXT_DEP" >> ./include/blasfeo_target.h
echo "#define EXT_DEP" >> ./include/blasfeo_target.h
echo "#endif"          >> ./include/blasfeo_target.h
echo "#ifndef OS_MAC" >> ./include/blasfeo_target.h
echo "#define OS_MAC" >> ./include/blasfeo_target.h
echo "#endif"         >> ./include/blasfeo_target.h
( cd auxiliary; /Applications/Xcode.app/Contents/Developer/usr/bin/make obj)
Parsing Makefile.rule
gcc -O2 -fPIC  -I/Users/jing/acados/external/blasfeo/include -DLA_HIGH_PERFORMANCE -DBLASFEO_REF_API -DMF_PANELMAJ -DK_MAX_STACK=300 -DUSE_C99_MATH -DEXT_DEP -DOS_MAC  -DEXTERNAL_BLAS_NONE -m64 -mavx2 -mfma -DTARGET_X64_INTEL_HASWELL   -c -o blasfeo_stdlib.o blasfeo_stdlib.c
clang: warning: argument unused during compilation: '-mavx2' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mfma' [-Wunused-command-line-argument]
gcc -O2 -fPIC  -I/Users/jing/acados/external/blasfeo/include -DLA_HIGH_PERFORMANCE -DBLASFEO_REF_API -DMF_PANELMAJ -DK_MAX_STACK=300 -DUSE_C99_MATH -DEXT_DEP -DOS_MAC  -DEXTERNAL_BLAS_NONE -m64 -mavx2 -mfma -DTARGET_X64_INTEL_HASWELL   -c -o blasfeo_processor_features.o blasfeo_processor_features.c
clang: warning: argument unused during compilation: '-mavx2' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mfma' [-Wunused-command-line-argument]
In file included from blasfeo_processor_features.c:46:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/14.0.3/include/cpuid.h:14:2: error: this header is for x86 only
#error this header is for x86 only
 ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/14.0.3/include/cpuid.h:287:5: error: invalid output constraint '=a' in asm
    __cpuid(__leaf, __eax, __ebx, __ecx, __edx);
    ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/14.0.3/include/cpuid.h:252:11: note: expanded from macro '__cpuid'
        : "=a"(__eax), "=r" (__ebx), "=c"(__ecx), "=d"(__edx) \
          ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/14.0.3/include/cpuid.h:302:5: error: invalid output constraint '=a' in asm
    __cpuid(__leaf, *__eax, *__ebx, *__ecx, *__edx);
    ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/14.0.3/include/cpuid.h:252:11: note: expanded from macro '__cpuid'
        : "=a"(__eax), "=r" (__ebx), "=c"(__ecx), "=d"(__edx) \
          ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/14.0.3/include/cpuid.h:316:5: error: invalid output constraint '=a' in asm
    __cpuid_count(__leaf, __subleaf, *__eax, *__ebx, *__ecx, *__edx);
    ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/14.0.3/include/cpuid.h:259:11: note: expanded from macro '__cpuid_count'
        : "=a"(__eax), "=r" (__ebx), "=c"(__ecx), "=d"(__edx) \
          ^
blasfeo_processor_features.c:183:5: error: invalid output constraint '=a' in asm
                : "=a" (reg_eax), "=b" (reg_ebx), "=c" (reg_ecx), "=d" (reg_edx)
                  ^
5 errors generated.
make[2]: *** [blasfeo_processor_features.o] Error 1
make[1]: *** [shared_library] Error 2
make: *** [blasfeo_shared] Error 2

FreyJo, 2023-10-05:
I think your cmake command misses the essential `..` at the end.
From the instructions:
`cmake -DACADOS_WITH_QPOASES=ON ..`

jy-cds, 2023-10-05:
Oh I see! I was able to run the `cmake -DACADOS_WITH_QPOASES=ON ..` command now! I thought the `..` was referring to the other arguments I could have passed on to this line... Sorry for my lack of experience with this type of operation!

After running the `cmake` line, when I run `make install -j4`, I received the following error:

> [ 29%] Building C object external/blasfeo/CMakeFiles/blasfeo.dir/auxiliary/timing.c.o
[ 29%] Building C object external/blasfeo/CMakeFiles/blasfeo.dir/auxiliary/v_aux_ext_dep_lib.c.o
[ 30%] Linking C shared library libblasfeo.dylib
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc  -std=c99 -Wall -fdiagnostics-show-option -Wno-error -w -DLA_HIGH_PERFORMANCE -DBLASFEO_REF_API -DMF_PANELMAJ -DK_MAX_STACK=300 -DUSE_C99_MATH -DEXT_DEP -DOS_MAC -DEXTERNAL_BLAS_NONE -O2 -fPIC -DTARGET_X64_INTEL_CORE -m64 -msse3 -O3 -DNDEBUG -arch x86_64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk -mmacosx-version-min=13.1 -dynamiclib -Wl,-headerpad_max_install_names -o libblasfeo.dylib -install_name /Users/jing/acados/build/external/blasfeo/libblasfeo.dylib CMakeFiles/blasfeo.dir/auxiliary/blasfeo_processor_features.c.o CMakeFiles/blasfeo.dir/auxiliary/blasfeo_stdlib.c.o CMakeFiles/blasfeo.dir/auxiliary/d_aux_common.c.o CMakeFiles/blasfeo.dir/auxiliary/memory.c.o CMakeFiles/blasfeo.dir/auxiliary/s_aux_common.c.o CMakeFiles/blasfeo.dir/kernel/generic/kernel_d_aux_lib.c.o CMakeFiles/blasfeo.dir/kernel/generic/kernel_daxpy_lib.c.o CMakeFiles/blasfeo.dir/kernel/generic/kernel_ddot_lib.c.o CMakeFiles/blasfeo.dir/kernel/generic/kernel_dgecp_lib4.c.o CMakeFiles/blasfeo.dir/kernel/generic/kernel_dgemm_4x4_lib4.c.o CMakeFiles/blasfeo.dir/kernel/generic/kernel_dgemm_diag_lib4.c.o CMakeFiles/blasfeo.dir/kernel/generic/kernel_dgemv_4_lib4.c.o CMakeFiles/blasfeo.dir/kernel/generic/kernel_dgeqrf_4_lib4.c.o CMakeFiles/blasfeo.dir/kernel/generic/kernel_dger_lib4.c.o CMakeFiles/blasfeo.dir/kernel/generic/kernel_dgetr_lib.c.o CMakeFiles/blasfeo.dir/kernel/generic/kernel_dgetr_lib4.c.o CMakeFiles/blasfeo.dir/kernel/generic/kernel_dgetrf_pivot_lib4.c.o CMakeFiles/blasfeo.dir/kernel/generic/kernel_dpack_buffer_lib4.c.o CMakeFiles/blasfeo.dir/kernel/generic/kernel_dpack_lib4.c.o CMakeFiles/blasfeo.dir/kernel/generic/kernel_dsymv_4_lib4.c.o CMakeFiles/blasfeo.dir/kernel/generic/kernel_saxpy_lib.c.o CMakeFiles/blasfeo.dir/kernel/generic/kernel_sdot_lib.c.o CMakeFiles/blasfeo.dir/kernel/generic/kernel_sgecp_lib4.c.o CMakeFiles/blasfeo.dir/kernel/generic/kernel_sgemm_4x4_lib4.c.o CMakeFiles/blasfeo.dir/kernel/generic/kernel_sgemm_diag_lib4.c.o CMakeFiles/blasfeo.dir/kernel/generic/kernel_sgemv_4_lib4.c.o CMakeFiles/blasfeo.dir/kernel/generic/kernel_sgetr_lib4.c.o CMakeFiles/blasfeo.dir/kernel/generic/kernel_sgetrf_pivot_lib4.c.o CMakeFiles/blasfeo.dir/kernel/generic/kernel_spack_lib4.c.o CMakeFiles/blasfeo.dir/kernel/generic/kernel_ssymv_4_lib4.c.o CMakeFiles/blasfeo.dir/kernel/sse3/kernel_align_x64.S.o CMakeFiles/blasfeo.dir/kernel/sse3/kernel_dgemm_4x4_lib4.S.o CMakeFiles/blasfeo.dir/kernel/sse3/kernel_dgemv_4_lib4.S.o CMakeFiles/blasfeo.dir/kernel/sse3/kernel_sgemm_4x4_lib4.S.o CMakeFiles/blasfeo.dir/auxiliary/d_aux_lib4.c.o CMakeFiles/blasfeo.dir/auxiliary/m_aux_lib44.c.o CMakeFiles/blasfeo.dir/auxiliary/s_aux_lib4.c.o CMakeFiles/blasfeo.dir/blasfeo_hp_pm/d_blas1_lib4.c.o CMakeFiles/blasfeo.dir/blasfeo_hp_pm/d_blas2_diag_lib.c.o CMakeFiles/blasfeo.dir/blasfeo_hp_pm/d_blas2_lib4.c.o CMakeFiles/blasfeo.dir/blasfeo_hp_pm/d_blas3_diag_lib4.c.o CMakeFiles/blasfeo.dir/blasfeo_hp_pm/d_blas3_lib4.c.o CMakeFiles/blasfeo.dir/blasfeo_hp_pm/d_lapack_lib4.c.o CMakeFiles/blasfeo.dir/blasfeo_hp_pm/s_blas1_lib4.c.o CMakeFiles/blasfeo.dir/blasfeo_hp_pm/s_blas2_diag_lib.c.o CMakeFiles/blasfeo.dir/blasfeo_hp_pm/s_blas2_lib4.c.o CMakeFiles/blasfeo.dir/blasfeo_hp_pm/s_blas3_diag_lib4.c.o CMakeFiles/blasfeo.dir/blasfeo_hp_pm/s_blas3_lib4.c.o CMakeFiles/blasfeo.dir/blasfeo_hp_pm/s_lapack_lib4.c.o CMakeFiles/blasfeo.dir/auxiliary/d_aux_ref.c.o CMakeFiles/blasfeo.dir/auxiliary/s_aux_ref.c.o CMakeFiles/blasfeo.dir/blasfeo_ref/d_blas1_ref.c.o CMakeFiles/blasfeo.dir/blasfeo_ref/d_blas2_diag_ref.c.o CMakeFiles/blasfeo.dir/blasfeo_ref/d_blas2_ref.c.o CMakeFiles/blasfeo.dir/blasfeo_ref/d_blas3_diag_ref.c.o CMakeFiles/blasfeo.dir/blasfeo_ref/d_blas3_ref.c.o CMakeFiles/blasfeo.dir/blasfeo_ref/d_lapack_ref.c.o CMakeFiles/blasfeo.dir/blasfeo_ref/s_blas1_ref.c.o CMakeFiles/blasfeo.dir/blasfeo_ref/s_blas2_diag_ref.c.o CMakeFiles/blasfeo.dir/blasfeo_ref/s_blas2_ref.c.o CMakeFiles/blasfeo.dir/blasfeo_ref/s_blas3_diag_ref.c.o CMakeFiles/blasfeo.dir/blasfeo_ref/s_blas3_ref.c.o CMakeFiles/blasfeo.dir/blasfeo_ref/s_lapack_ref.c.o CMakeFiles/blasfeo.dir/auxiliary/d_aux_ext_dep.c.o CMakeFiles/blasfeo.dir/auxiliary/d_aux_ext_dep_common.c.o CMakeFiles/blasfeo.dir/auxiliary/i_aux_ext_dep_lib.c.o CMakeFiles/blasfeo.dir/auxiliary/s_aux_ext_dep.c.o CMakeFiles/blasfeo.dir/auxiliary/s_aux_ext_dep_common.c.o CMakeFiles/blasfeo.dir/auxiliary/timing.c.o CMakeFiles/blasfeo.dir/auxiliary/v_aux_ext_dep_lib.c.o 
[ 30%] Built target blasfeo
make: *** [all] Error 2

Does this mean I still had issues at the previous `cmake` line? Or this was a problem with the BLASFEO target (for which in the `Makerfile.rule`, I commented out on all other lines and left `BLASFEO_TARGET = ARMV8A_APPLE_M1`)?

giaf, 2023-10-06:
In both make and cmake you seem to be using x86 targets (respectively Haswell and Core), while you should use the Apple M1 one (or if not available the ARM Cortex A57 that shares the same code).

With this commit https://github.com/giaf/blasfeo/commit/b39df4d6774dcf37aa759ea65a26f2c8affff281 Apple M1 is supported in the BLASFEO cmake too.

jy-cds, 2023-10-12:
Apologies for the late response - I was out of the office for a few days. 

I restarted a new virtual environment and followed the installation instruction from the start. Now I get 
> CMake Error at external/blasfeo/CMakeLists.txt:166 (message):
  Target ARMV8A_APPLE_M1 is not supported
during the step `cmake -DACADOS_WITH_QPOASES=ON ..`. I suspect this is because when I was running the previous line `git submodule update --recursive --init`, the latest version of `blasfeo` was not included? 

Can you point me to or show me how to update to the latest `blasfeo`? I thought it was already installed as a package, but I realized it was not. Furthermore, `acados` seems to link `blasfeo` into its path. 

Thanks!!
Jing

Tim-Salzmann, 2023-10-12:
Hi Jing,

here is my process to compile Acados for M1.

1. Make sure cmake is an arm64 binary and not a Rosetta amd_x64
`file $(where cmake)` --> `Mach-O 64-bit executable arm64`.
2. 
```
git clone https://github.com/acados/acados.git
cd acados
git submodule update --recursive --init
```

3. For the current main branch of qpoases I had to do a slight adaptation in `external/qpoases/CMakeLists.txt` to get it to work with M1 (not sure if this is still necessary).

```
-    set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -w -fPIC")
+    set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -w -fPIC -Wno-error=implicit-function-declaration")
```

4. Proceed with normal CMake and install:
```
mkdir -p build
cd build
cmake -DACADOS_WITH_QPOASES=ON ..
make install -j4
```

Best
Tim

jy-cds, 2023-10-12:
Thanks Tim! deleting and adding the line with `set(CMAKE)` helped resolve all the issues regarding installation! I then followed through the rest of the installation as well as `python interface` installation, and everything seemed successfully installed. However, when I run the `minimal_example_ocp.py` to test the installation, I came across the following warning and error:

1.  Warning
> Warning: Did not find environment variable ACADOS_SOURCE_DIR, guessed ACADOS_PATH to be /Users/jing/acados.
Please export ACADOS_SOURCE_DIR to avoid this warning.

I had already performed the instructed `export ACADOS_SOURCE_DIR="
/Users/jing/acados"` but that seemed to have done nothing. I wonder if setting the `export` line had no effect for some reason. 

2. Error (may be related to the warning)

> cc -fPIC -std=c99   -O2 -I/Users/jing/acados/include -I/Users/jing/acados/include/acados -I/Users/jing/acados/include/blasfeo/include -I/Users/jing/acados/include/hpipm/include  -c -o acados_solver_pendulum_ode.o acados_solver_pendulum_ode.c
acados_solver_pendulum_ode.c:37:10: fatal error: 'acados_c/ocp_nlp_interface.h' file not found
#include "acados_c/ocp_nlp_interface.h"
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make: *** [acados_solver_pendulum_ode.o] Error 1
Traceback (most recent call last):
  File "/Users/jing/acados/examples/acados_python/getting_started/minimal_example_ocp.py", line 104, in <module>
    main()
  File "/Users/jing/acados/examples/acados_python/getting_started/minimal_example_ocp.py", line 83, in main
    ocp_solver = AcadosOcpSolver(ocp, json_file = 'acados_ocp.json')
  File "/Users/jing/acados/interfaces/acados_template/acados_template/acados_ocp_solver.py", line 972, in __init__
    self.__acados_lib = CDLL(libacados_filepath)
  File "/Users/jing/opt/anaconda3/envs/building_mpc/lib/python3.9/ctypes/__init__.py", line 374, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: dlopen(/Users/jing/acados/lib/libacados.dylib, 0x0006): tried: '/Users/jing/acados/lib/libacados.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Users/jing/acados/lib/libacados.dylib' (no such file), '/Users/jing/acados/lib/libacados.dylib' (no such file)

It looked like I was missing some library file? I wonder if it is because the previous `export` steps did not actually take effect. 

Thanks again!
Jing

jy-cds, 2023-10-13:
HI Tim,

Can you explain more about "while running a python optimization"? I followed [4] and tried to put these lines in the terminal before but that did not have an effect. After a quick search, it seems that this type of lines should be put into a `.bashrc` file? If so, which file should I have put the lines? Thanks!

Tim-Salzmann, 2023-10-13:
Exporting the variables in the terminal before running should be sufficient. Just to double check: You did replace `<acados_root>` with the path to your acados git clone? E.g. I would do
```
export ACADOS_SOURCE_DIR=/Users/Tim/Documents/Study/PhD/Code/Tools/acados
export DYLD_LIBRARY_PATH=/Users/Tim/Documents/Study/PhD/Code/Tools/acados/lib
```

jy-cds, 2023-10-13:
Hi Tim,

Yes I did replace the `<acados_root>` with my acados root file location (similar to yours). After running these two lines in the terminal, and running the `minimal_example_ocp.py`, I have the exact same (1) warning and (2)error as mentioned above, and the terminal `export` lines seem to have done nothing. I should note that I run the `minimal_example_ocp.py` from VScode. 

Best,
Jing

Tim-Salzmann, 2023-10-13:
Hi Jing,

> I should note that I run the minimal_example_ocp.py from VScode.
The environment variables must be set in the same environment where the python script is run. VSCode will run the script in a separate environment. Thus, setting the variables in a different terminal will have no effect. To add the variables to the VSCode python environment, you must adapt the run config [1].

>  I do not see a lib folder in my Users/jing/acados
This folder should have been created during the successful execution of `make install -j4` [2].

Best
Tim

[1] https://code.visualstudio.com/docs/python/settings-reference
[2] https://docs.acados.org/installation/index.html

jy-cds, 2023-10-13:
Hi Tim, Gianluca, and Jonathan, 

After deleting the `build` folder and re-run the `cmake` and `make install -j4` command again, as well as setting the PATH variables in VScode, I successfully ran the ocp example!! Thank you so so much for all your help and your patience!

Best,
Jing

closed 2023-10-13 by jy-cds