acados/acados #973: How to use customized cost function in acados similiar to ACADO

fayyazchemban, 2023-10-11:
In ACADO, the cost function is added like below:

Function rf, rfN;
and then adding rf << ... 
in which you can combine various states or intermediate states.

How can I do the same in acados ?

FreyJo, 2023-10-12:
You can use NONLINEAR least squares or EXTERNAL cost and define any cost term using CasADi symbolics.

pockerbot, 2023-10-26:
Thanks for the answer. If there is an example i can refer to in the acados with C interface, please let me know. Also, Is the cost something which is fixed in the model.py if i am using python for code generation or is there a way i can change the cost of each reference on the fly in C++ based on some scenarios. like if i am understanding correctly, ocp_nlp_cost_model_set(nlp_config, nlp_dims, nlp_in, i, "yref", yref); is used to set the reference, similarly how can i change the cost of those references in the fly in C++ if it is using auto generated code interface.

pockerbot, 2023-11-13:
Is it possible to use the api ocp_nlp_cost_model_set(nlp_config, nlp_dims, nlp_in, i, "W", W) on fly to set a new cost for tuning a system. Like I am using acados with ROS and I am trying to tune a robot for trajectory tracking. I am using dynamic reconfigure to change the cost and in the callback i am using ocp_nlp_cost_model_set(nlp_config, nlp_dims, nlp_in, i, "W", W) to change the cost on the fly. At times i am getting weird behaviour. Is it something wrong to do like that ?

FreyJo, 2023-11-13:
> if i am understanding correctly, ocp_nlp_cost_model_set(nlp_config, nlp_dims, nlp_in, i, "yref", yref);
> i am using ocp_nlp_cost_model_set(nlp_config, nlp_dims, nlp_in, i, "W", W) to change the cost on the fly

Both are correct. I don't know if your solver converges or if your solution just is not what you expect somehow.
You can also make a parametric cost and update the parameters to change a more flexible cost function on the fly, especially with the EXTERNAL cost module.

closed 2023-11-13 by FreyJo