acados/acados #981: Invalid MEX-file when running MATLAB examples

KarasevPavel, 2023-10-26:
Hello! I have a problem with a warning:
```
Warning: The following error was caught while executing 'pendulum_mex_solver' class destructor:
Error using cd
Path must be a text scalar.

Error in pendulum_mex_solver/delete (line 64)
            cd(obj.code_gen_dir);

Error in pendulum_mex_solver (line 49)
            make_mex_pendulum();

Error in acados_ocp (line 209)
            obj.t_ocp = mex_solver();

Error in minimal_example_closed_loop (line 125)
ocp = acados_ocp(ocp_model, ocp_opts); 
> In pendulum_mex_solver (line 49)
In acados_ocp (line 209)
In minimal_example_closed_loop (line 125) 
```
and than an error:
```
Invalid MEX-file '/home/pavel/acados/examples/acados_matlab_octave/getting_started/c_generated_code/acados_mex_create_pendulum.mexa64': libacados_ocp_solver_pendulum.so: cannot open shared object file: No such file or directory

Error in pendulum_mex_solver (line 50)
            obj.C_ocp = acados_mex_create_pendulum();

Error in acados_ocp (line 209)
            obj.t_ocp = mex_solver();

Error in minimal_example_closed_loop (line 125)
ocp = acados_ocp(ocp_model, ocp_opts);
```
when I run examples from the folder `acados_matlab_octave` from this repository

My system:

Processor: intel core i7-9750H
OS: Ubuntu 22.04
MATLAB: 2022a (2021b the same result)
CasADi: 3.4.0 (as mentioned in the **Matlab + Simulink and Octave Interface** guide)

I have done the installation procedure from the links:
1. [Installation](https://docs.acados.org/installation/index.html)
2. [Matlab + Simulink and Octave Interface](https://docs.acados.org/matlab_octave_interface/index.html)

I did some workaround and my temporary solution now is to add a path to the `c_generated_code` folder, which appears in the Matlab running directory to the `LD_LIBRARY_PATH` variable before launching MATLAB. May be we need to add a path to the `c_generated_code` folder in all `env.sh` scripts of all examples? But I think it is not a good idea

Regards, Pavel.

FreyJo, 2023-10-26:
Hi Pavel,

I am not working with Maltab these days.
But I think that after the latest changes to the Matlab interfaces, see #934, and especially if the model lib is in `c_generated_code`, this folder should indeed be in `LD_LIBRARY_PATH`.

I tried to clarify here that CasADi 3.4 is not a requirement:
https://github.com/acados/acados/pull/980/commits/95056fb6f1d0759dc51bd62ba7adab61c39b98ef

Best,
Jonathan

FreyJo, 2023-11-07:
Hi Pavel,

I just tested the example you mentioned on my machine and it seems to work without explicitly adding `c_generated_code` to the LD_LIBRARY_PATH. Did you run the example from the folder it is located in?

Also, I addressed the CasADi requirement here:
https://github.com/acados/acados/pull/991

frnyb, 2023-11-21:
I am experiencing the same issue. When following the documentation, running any example (in this case `linear_mass_spring_model`) I get:
```
Warning: The following error was caught while executing 'sim_model_mex_sim_solver' class destructor:
Error using cd
Path must be a text scalar.

Error in sim_model_mex_sim_solver/delete (line 79)
            cd(obj.code_gen_dir);

Error in sim_model_mex_sim_solver (line 46)
            make_mex_sim_sim_model();

Error in acados_sim (line 110)
            obj.t_sim = mex_sim_solver();

Error in example_sim (line 112)
sim = acados_sim(sim_model, sim_opts); 
> In sim_model_mex_sim_solver (line 46)
In acados_sim (line 110)
In example_sim (line 112) 
Invalid MEX-file
'/home/ffn/acados/examples/acados_matlab_octave/linear_mass_spring_model/c_generated_code/acados_sim_create_sim_model.mexa64':
libacados_sim_solver_sim_model.so: cannot open shared object file: No such file or directory

Error in sim_model_mex_sim_solver (line 47)
            obj.C_sim = acados_sim_create_sim_model();

Error in acados_sim (line 110)
            obj.t_sim = mex_sim_solver();

Error in example_sim (line 112)
sim = acados_sim(sim_model, sim_opts);
```

> Did you run the example from the folder it is located in?

Yes, I do run it from the folder of the example, ´<acados>/examples/acados_matlab_octave/linear_mass_spring_model´.

When adding the `c_code_gen` dir to `LD_LIBRARY_PATH` in `env.sh`, I get 
```
Unrecognized method, property, or field 'C_sim' for class 'acados_sim'.

Error in example_sim (line 115)
sim.C_sim
```

Can anybody help shed some light on this? Any help is appreciated.

frnyb, 2023-11-21:
Adding the `c_code_gen` dir to `env.sh` is done in this PR: https://github.com/acados/acados/pull/996
However, it still leaves me with 
```
Unrecognized method, property, or field 'C_sim' for class 'acados_sim'.

Error in example_sim (line 115)
sim.C_sim
```

FreyJo, 2023-11-21:
Addressed error reported by @frnyb in https://github.com/acados/acados/pull/996 too.
I hope also all Matlab issues of @KarasevPavel are fixed after the merge.

closed 2024-05-06 by sandmaennchen