acados/acados #985: Python interface installation issues

Azircarry, 2023-10-29:
I'm installing the python interface of acados in ubuntu18.04 system, python version is 3.6, when I installed according to the official document, there was an error in the following command, which caused the download of acados_template package to fail, but when I downloaded other packages, it can be downloaded successfully, can you help me，thanks

When I execute the following command：
pip3 install acados_template -i http://pypi.douban.com/simple/ --trusted-host pypi.douban.com

The following error will occur：
Collecting acados_template
  Could not find a version that satisfies the requirement acados_template (from versions: )
No matching distribution found for acados_template
[image]

FreyJo, 2023-10-30:
Sorry, you did not follow the installation instructions of the Python interface properly.
https://docs.acados.org/python_interface/index.html#installation

closed 2023-10-30 by FreyJo

Azircarry, 2023-10-31:
Hello, thanks for your reply, but even if I install according to the installation instructions, I still get the following error, I don't know what the reason is, in addition, I went to the python third-party package official website to check, and I didn't find acados_template
pack，Sorry to bother you
[image]

Azircarry, 2023-10-31:
Thanks for your reply？It is ok

Azircarry, 2023-10-31:
Thanks for your reply!  It is ok。