acados/acados #986: The python interface example runs incorrectly

Azircarry, 2023-10-31:
Hello, I have installed the python interface, but when I run the minimal_example_ocp.py, I get the following error, can you help me, thank you 

[image]

FreyJo, 2023-10-31:
Which python version are you using?
It seems to be an fstring problem.
It surely works with Python>=3.8.
I guess we should test with something lower too or add this requirement explicitly.

Azircarry, 2023-11-01:
python3.6.9

Azircarry, 2023-11-01:
> Which python version are you using? It seems to be an fstring problem. It surely works with Python>=3.8. I guess we should test with something lower too or add this requirement explicitly.

I installed ROS, PX4, gazebo before doing the acados install and python example tests and don't know if this will have an impact

Azircarry, 2023-11-01:
> Which python version are you using?

python3.6.9

Azircarry, 2023-11-01:
Can I just download python 3.8 and test it without reinstalling the acados and python interfaces

FreyJo, 2023-11-01:
You will need to install the Python interface again, i.e. do `pip install -e <path_to_template>`.

Azircarry, 2023-11-01:
> You will need to install the Python interface again, i.e. do `pip install -e <path_to_template>`.

After the operation, the following interface appears, but I don't know if it is successful. Please accept my heartfelt thanks.
[image]

Azircarry, 2023-11-02:
> Which python version are you using? It seems to be an fstring problem. It surely works with Python>=3.8. I guess we should test with something lower too or add this requirement explicitly.

I really wish I could run on python3.6 as I have some projects done with python3.6, but I saw in previous posts that someone did run the example successfully with python3.6, thanks of course for your help

FreyJo, 2023-12-18:
I cannot easily test with Python 3.6, but 3.7 works after merging https://github.com/acados/acados/pull/1006
Thus, I am closing this.

closed 2023-12-18 by FreyJo

Azircarry, 2023-12-18:
> 我无法轻松使用 Python 3.6 进行测试，但 3.7 在合并后有效 #1006 因此，我正在关闭它。

Ok  ！Thank you again for your help and reply