acados/acados #987: Running extensive_example_ocp.m on MATLAB linux

Ajin2305, 2023-10-31:
After the installation of acados i was running the examples in the folder acados/examples/acados_matlab_octave/getting_started

I managed to run most of the examples (minimal ones) by clicking run, except extensive_example_ocp.  (Using MATLAB R2020b)

The error message was as follows
[image]

Kindly let me know if I am doing something wrong.

But, with the following modification in the file detect_dims_ocp.m. I managed to successfully run the example
`            if(all(opts.sim_method_num_stages) == 1 || all(opts.sim_method_num_stages) == 2 || ...
                all(opts.sim_method_num_stages) == 3 || all(opts.sim_method_num_stages) == 4)`
                
If the following change is acceptable let us try to update this in the code such that other users will be able to run the example without any issues, thanks.

Ajin2305, 2023-10-31:
Output after the changes above were carried out. 
[image]

FreyJo, 2023-10-31:
I think the change you described makes sense, you can open a PR.

Ajin2305, 2023-10-31:
Hi @FreyJo  please find the pull request here - https://github.com/acados/acados/pull/989

closed 2023-11-02 by FreyJo