acados/acados #999: acados_tic when acados_timer is real_t

pockerbot, 2023-11-22:
When the acados_timer in timing.h is dummy type, the acados_tic changes value in memory block larger than the real_t size changing values of adjacent memory block as well.

FreyJo, 2023-11-28:
How did you find that?
What kind of setup are you using?
How can this be reproduced?
If you can propose a fix via pull request, it would be highly appreciated.

pockerbot, 2023-12-05:
I had a class with the following member variables close to each other like:
```
acados_timer m_timer;
double m_test;
```
Then when in the src file, i did acados_tic(&m_timer) then the value of m_test also got changed. I shifted to using c++ timer now for time calculation for now.

FreyJo, 2023-12-15:
I reproduced the issue and addressed it in https://github.com/acados/acados/pull/1005
More specifically, the issue was that `MEASURE_TIMINGS` is defined when compiling the acados shared library, but when one links against it and compiles without `-DMEASURE_TIMINGS`, the timer has not enough memory.