acados/acados #1002: Use acados with MATLAB in Windows

DoneZai, 2023-12-02:
Hello everyone! I was tring to use acados in window 10 with matlab 2022b, and I followed the document to install the acados normally. But when I tried to use the minimal_example_ocp.m as said in tutorial, the error comes like

> MEX file
> 'D:\Code\acados\examples\acados_matlab_octave\getting_started\c_generated_code\acados_mex_create_pendulum.mexw64'  The specified module could not be found.**
 
Since before I met similar problem in ubuntu, and solved it by add ./c_generated_code in env.sh, I tried it also in windows although LD_LIBRARY_PATH didn't make sense in Windows. I also add ./c_generated to PATH by windows GUI.  But that error is still there :( 
Thanks in advance.

DoneZai, 2023-12-02:
By the way, .\acados\examples\acados_matlab_octave\getting_started\c_generated_code\make_mex_pendulum.m, line 34 should be     

>     opts.output_dir = pwd();

instead of 

>     opts.output_dir = pwd;

FreyJo, 2023-12-15:
I don't see why this line is a problem?

> `opts.output_dir = pwd;`

Unfortunately, I don't know the reason of your actual issue.