acados/acados #1017: T_renderer issue while using python interface

Y1bo-Zhao, 2024-01-11:
I am trying to use acados with the Python interface and the t_renderer file is not being detected for some reason.

There was another thread that suggested building the binaries natively, which I tried but was still unable to rectify the problem.

This is the error message that shows up when I try running the example : ‘minimal_example_ocp.py’
[image]

I also noticed that even though the download of the t_renderer file is initiated there is not confirmation message of it’s completion.
As you can see the error message reads No such file or directory: ‘~/acados/bin/t_renderer’ which is not true as that is where I copied the t_renderer from the built git repo to.
Any help would be appreciated. :smiley:

HuiQianPeng0401, 2024-10-24:
I also encountered the same issue. Have you resolved it?

josipkh, 2025-07-18:
Seems related to https://github.com/acados/tera_renderer/issues/10, which was fixed in https://github.com/acados/acados/pull/1561.
Feel free to reopen if needed.