acados/acados #1019: Small change to HPIPM API

giaf, 2024-01-14:
There was a small change in the HPIPM API for some setters related to x0 elimination https://github.com/giaf/hpipm/commit/2b18b0f419046cb00806520e2cf4bd691d149277

Next time HPIPM is update in acados, there will need to be made a small change to the sources
```acados/ocp_qp/ocp_qp_partial_condensing.c```
```acados/ocp_qp/ocp_qp_full_condensing.c```

FreyJo, 2024-01-15:
Thanks for the heads-up!
Although I honestly don't see which convention was fixed with that change in HPIPM.

giaf, 2024-01-15:
All other setters in HPIPM have the value before and the struct as last argument.
Also the value is passed by reference, so there can be a global setter using a field on top of the tailored setters, making use in python or matlab interfaces easier.

closed 2024-01-15 by FreyJo