acados/acados #1022: OSQP condensing solvers violate inequality/polytopic constraints

FranekStark, 2024-01-16:
I use the C `ocp_qp_interface` and have a relative simple optimization problem that, among the dynamics and some bounding box constraints, constraints the following input inequality/polytopic constraints for each stage:

```
EigenTypeC C =  {1, 0, -mu,
                 1, 0,  mu,
                 0, 1, -mu,
                 0, 1,  mu};

EigenTypec c_lb = {NEG_UNBOUND,
                   0,
                   NEG_UNBOUND,
                   0}
EigenTypec c_ub = {0,
                   POS_UNBOUND,
                   0,
                   POS_UNBOUND}
```
Where `mu=0.49` and `-NEG_UNBOUND = POS_UNBOUND = 9999999999`.
(The abs of the first and second input can be only half as big as the first input)

I set for each stage `k` I set the dimensions and afterwords the data:
```
ocp_qp_in_set(solver_config_, qp_in_, k, "D", C.data());
ocp_qp_in_set(solver_config_, qp_in_, k, "lg", c_lb.data());
ocp_qp_in_set(solver_config_, qp_in_, k, "ug", c_ub.data());
```

When now solving the problem using the partial condensed methods `PARTIAL_CONDENSING_OSQP` and `PARTIAL_CONDENSING_HPIPM`, this constraints are respected. I get for both very similar results:  (input of first stage)
`(27.69, -0.01, 61.45)` and `(27.61, -0.01, 61.36) `

However, using fully condensed solvers I get:
with  `FULL_CONDENSING_QPOASES`: `(-200.00, 8.22, -0.00)`
with  `FULL_CONDENSING_HPIPM`: ` (-200.00, 8.15, 0.00)`

The constraints are clearly violated.
Do I miss something in the problem setup, or is this a bug in the condensing method?

FreyJo, 2024-01-16:
Hi,
thanks for reporting this!
Did you check the status, do the solvers report to have run successfully?
Condensing can worsen the conditioning of the problem. What is your partial condensing horizon?
Maybe you can try different horizons? A partial condensing horizon of 0 or 1 should be similar to full condensing.

FranekStark, 2024-01-16:
@FreyJo The status for both cases is 'ACADOS_SUCCESS'. For qpOASES i checked the actual return of the solver which is also 0.
Regarding the partial condensing horizon, I was using the default setting (as far as I understand that means no condensing at all - please correct me if I am wrong). However, with the partial solvers, even for a horizon of 1, I don't see these violations (For a horizon of 0 I get an exception).

One question out of interest. Why can condensing have an influence into the inequality constraints of the input? Shouldn't the matrices just stack up?

sandmaennchen, 2024-01-17:
Hi @FranekStark, this seems indeed weird. Could you provide a minimal working example that we could run? Thanks a lot!

FreyJo, 2024-01-29:
Any update on this, after updating OSQP?
https://github.com/acados/acados/pull/1033

FranekStark, 2024-01-29:
Hi @FreyJo haven't found the time to investigate that more. As this only occurs in very 'constructed' situations and in never saw it in reality yet, it's for now not too bad. However, it is still on my list to provide a small example where it happens.