acados/acados #1032: Update solvers

FranekStark, 2024-01-29:
The solver versions shipped with ACADOS are rather outdated.
In the meantime, some critical bugs have been fixed, e.g. https://github.com/robotology/osqp-eigen/issues/81.
Unfortunately, just checking out the newest version does not work, as also the interfaces have slightly changed. Hence, the solver interfacing has to be updated too.

FreyJo, 2024-01-29:
I see, it definitely makes sense to update OSQP.
HPIPM is up to date, DAQP is very recent and the C implementation of  qpOASES is not acitvely developed.

imciner2, 2024-01-29:
>  Unfortunately, just checking out the newest version does not work, as also the interfaces have slightly changed. Hence, the solver interfacing has to be updated too.

Don't just do a git checkout on the master branch for this. The master branch of OSQP is currently a 1.0 beta version that is still being finalized and shouldn't be used in production environments just yet. The current 0.6.3 version can be found under the GitHub releases: https://github.com/osqp/osqp/releases/tag/v0.6.3. It should be very close to the same API as exists in Acados already, but I can't recall if any of the data structure layouts changed between the versions (I think there may have been a few small changes).

FranekStark, 2024-01-29:
@imciner2 yes I actually checked out the OSQP v 0.6.3 and as you said it is only minor changes in the interface. On my version it works after minor changes. I can prepare a pull request.

closed 2024-01-29 by FreyJo