acados/acados #1041: TypeError: __init__() got an unexpected keyword argument 'winmode

james1kobby, 2024-02-27:
Traceback (most recent call last):
  File "main.py", line 39, in <module>
    run_nominal_control(chain_params)
  File "/home/ubuntu/Code_Study/acados/examples/acados_python/chain_mass/run_nominal_control.py", line 196, in run_nominal_control
    acados_integrator = export_chain_mass_integrator(n_mass, m, D, L)
  File "/home/ubuntu/Code_Study/acados/examples/acados_python/chain_mass/export_chain_mass_integrator.py", line 78, in export_chain_mass_integrator
    acados_integrator = AcadosSimSolver(sim, json_file = 'acados_ocp_' + model.name + '.json')
  File "/home/ubuntu/Code_Study/acados/interfaces/acados_template/acados_template/acados_sim_solver.py", line 272, in __init__
    self.__acados_lib = DllLoader(libacados_filepath, winmode=self.winmode)
TypeError: __init__() got an unexpected keyword argument 'winmode

james1kobby, 2024-02-28:
File "/home/ubuntu/Code_Study/acados/interfaces/acados_template/acados_template/acados_sim_solver.py", line 272, in init
self.__acados_lib = DllLoader(libacados_filepath, winmode=self.winmode)

修改成
File "/home/ubuntu/Code_Study/acados/interfaces/acados_template/acados_template/acados_sim_solver.py", line 272, in init
self.__acados_lib = DllLoader(libacados_filepath)

因为winmode=self.winmode时windows系统上加载动态库的标志

FreyJo, 2024-02-28:
what version of ctypes do you have?

james1kobby, 2024-02-29:
@FreyJo ，I don't know, how to updating ctypes？ I use python virtualenv to install ctypes.  
can yu tell me? thank you  very much.

FreyJo, 2024-03-14:
I am not sure. I think the ctypes version needed is not that recent.
What python version do you use? Could just update that?

FreyJo, 2024-05-08:
I fixed this in https://github.com/acados/acados/pull/1101

closed 2024-05-08 by FreyJo