acados/acados #1042: Non-linear constraints cause a crash on v0.3.0

parakhm95, 2024-02-28:
Our nonlinear constraints specified by 
```
    for (int i = 0; i < N; i++) {
      // nonlinear constraints for stages 0 to N-1
      ocp_nlp_constraints_model_set(nlp_config, nlp_dims, nlp_in, i, "nl_constr_h_fun_jac", &acados_ocp_capsule->nl_constr_h_fun_jac[i]);
      ocp_nlp_constraints_model_set(nlp_config, nlp_dims, nlp_in, i, "nl_constr_h_fun", &acados_ocp_capsule->nl_constr_h_fun[i]);

      ocp_nlp_constraints_model_set(nlp_config, nlp_dims, nlp_in, i, "lh", lh);
      ocp_nlp_constraints_model_set(nlp_config, nlp_dims, nlp_in, i, "uh", uh);
    }
```
cause a crash of our controller when we use acados v0.3.0 and only work upto the v0.2.6. Has something changed and is this method not correct anymore? How can we help you debug this? 

Thank you.

parakhm95, 2024-02-28:
@penickar @prochazo Maybe this answers that question through #994?

parakhm95, 2024-02-28:
@FreyJo  We create a model for our dynamics through a python interface and then use a c++ wrapper to modify the generated C code such that we do change the params, initial state, control reference, yref through our C++ wrapper.

FreyJo, 2024-02-28:
If you specify the constraints in python as you stated, it is only defined for the intermediate shooting nodes 1 to N-1, before the commit I pointed to from 0 to N-1.
If you want to also specify it at the initial shooting node, you have to use the new properties, `con_h_expr_0`, `lh_0`, `uh_0`.

parakhm95, 2024-03-03:
Our testing points to some other change causing this because setting the value of i to start from 1 in the for loop of the cpp file did not solve the issue, and the lines had to be commented out entirely. Has the function call changed in some way? 

This would break the non-linear constraints altogether for any index.

FreyJo, 2024-03-04:
I am not aware of any other breaking change introduced by this release. Maybe you narrow down the acados commit by doing a git bisect.

parakhm95, 2024-03-28:
Found it, it is the commit a778a0526399fff69bd52049aabdd5b3010a7da1

FreyJo, 2024-03-28:
Ok, this is the one I expected / told you about.

closed 2024-04-08 by FreyJo

parakhm95, 2024-04-08:
Hi, did you make a change to solve this?

FreyJo, 2024-04-18:
You narrowed it down to this `acados` commit [a778a05](https://github.com/acados/acados/commit/a778a0526399fff69bd52049aabdd5b3010a7da1)

> Found it, it is the commit [a778a05](https://github.com/acados/acados/commit/a778a0526399fff69bd52049aabdd5b3010a7da1)

And you say that it is a change in the C interface of acados.
However, you can easily verify that this commit does only change 2 lines in the C part of acados, namely commenting a print statement.

So, it needs to be related to your use of the Python interface and the generated code.