acados/acados #1056: nonlinear constraint slack variable at stage 0！

GuoZhongYi-Student, 2024-03-24:
I recently met a problem with acados in simulink. I have a nonlinear constraint and I would like to add a stage 0 slack variable to this nonlinear constraint, but I didn't find where Jsbh_0 is set, so I would like to ask how I should add a stage 0 relaxation variable.

FreyJo, 2024-03-26:
Hi,
I implemented it here:
https://github.com/acados/acados/pull/1059

GuoZhongYi-Student, 2024-03-27:
Thanks for such a quick reply, I hope to use it for the next version!

closed 2024-03-27 by FreyJo

GuoZhongYi-Student, 2024-03-27:
When I set the constr_Jsh_0 option with a new version of acados, the following error is reported:

```
CMakeFiles/ocp_pathtracking_model.dir/acados_solver_pathtracking_model.c.obj
C:\Users\gzy\Desktop\OCPSolverToolbox\acadosv3.1\acados\examples\acados_matlab_octave\obs_avoid\c_generated_code\acados_solver_pathtracking_model.c:`
In function 'pathtracking_model_acados_create_5_set_nlp_in':
C:\Users\gzy\Desktop\OCPSolverToolbox\acadosv3.1\acados\examples\acados_matlab_octave\obs_avoid\c_generated_code\acados_solver_pathtracking_model.c:669:18:
error: expected expression before '[' token
     idxsh_0[0] = [2];
                  ^
C:\Users\gzy\Desktop\OCPSolverToolbox\acadosv3.1\acados\examples\acados_matlab_octave\obs_avoid\c_generated_code\acados_solver_pathtracking_model.c:670:18:
error: expected expression before '[' token
     idxsh_0[1] = [3];
                  ^
CMakeFiles\ocp_pathtracking_model.dir\build.make:135: recipe for target
'CMakeFiles/ocp_pathtracking_model.dir/acados_solver_pathtracking_model.c.obj' failed
mingw32-make.exe[2]: ***
[CMakeFiles/ocp_pathtracking_model.dir/acados_solver_pathtracking_model.c.obj] Error 1
CMakeFiles\Makefile2:111: recipe for target 'CMakeFiles/ocp_pathtracking_model.dir/all' failed
mingw32-make.exe[1]: *** [CMakeFiles/ocp_pathtracking_model.dir/all] Error 2
Makefile:134: recipe for target 'all' failed
mingw32-make.exe: *** [all] Error 2
error: ocp_generate_c_code (line 281)
    acados_template_mex.compile_ocp_shared_lib(obj.acados_ocp_nlp_json.code_export_directory)
error: acados_ocp (line 203)
            ocp_generate_c_code(obj);
error: pathtracking_ocp (line 178 )
ocp = acados_ocp(ocp_model, ocp_opts);
```

note that I have four nonlinear constraints and only need to soften the last two, the corresponding program is:

```
nh       = 4;
nsh      = 2;
Jsh      = zeros(nh,nsh);
Jsh(3,1) = 1;
Jsh(4,2) = 1;
ocp_model.set('constr_Jsh_0', Jsh);
% L1 slack (linear term)
ocp_model.set('cost_zl_0', 100 * ones(nsh,1));
ocp_model.set('cost_zu_0', 100 * ones(nsh,1));
% L2 slack (squared term)
ocp_model.set('cost_Zl_0', 100 * ones(nsh,nsh));
ocp_model.set('cost_Zu_0', 100 * ones(nsh,nsh));
```

When I changed constr_Jsh_0 to constr_Jsh, cost_zl_0 to cost_zl, the error disappeared.