acados/acados #1075: Experiencing broken solver in 0.3.1

Veng97, 2024-04-17:
I have been using Acados for over a year for path optimization and tracking, but suddenly (without any changes to my code) the solver has become unstable.
I have just updated my local build tool chain to test with Acados 0.2.5, 0.2.6 and 0.3.0 - they were all stable. From 0.3.1 my problem is no longer solvable and it results in the error: ACADOS_MAXITER.

Settings:
ocp.solver_options.qp_solver = "FULL_CONDENSING_HPIPM"
ocp.solver_options.hessian_approx = "GAUSS_NEWTON"
ocp.solver_options.integrator_type = "ERK"
ocp.solver_options.print_level = 0
ocp.solver_options.qp_solver_warm_start = 1
ocp.solver_options.nlp_solver_max_iter = 500
ocp.solver_options.nlp_solver_type = "SQP"

Context:
I'm using the python Acados library to generate the solver, them i'm calling the Acados functions from a C++ application.
From my perspective there could be a change in the default options of the solver or maybe in the C++ interface, but nothing i could see in the 'diff' of the generated c-interface files.

I'm sorry I can't provide more details about the problem, but I have no idea where to look. I'm posting this here in hope I'm not the only one who has experienced this.

FreyJo, 2024-04-18:
Hi,

I am not aware of any breaking changes in acados 0.3.1 compared to 0.3.0.
I suggest to do a `git bisect` through the changes between those two versions and narrow it down to one of the 16 commits:
https://github.com/acados/acados/compare/v0.3.0...v0.3.1

Let me know if you find anything!

Veng97, 2024-04-18:
I took the commits from the top and hit the one that introduced an error on my end.
PASSES: e84fa19cb12db0996dc29cb039121dc70e93a039 
PASSES: 7cf7a3af67467038d868b95536e4a1623d1c3325 
FAILS: 37e17d31890ab54e5a855f1fe787fbf2f5d43bdb 
(https://github.com/acados/acados/commit/37e17d31890ab54e5a855f1fe787fbf2f5d43bdb)

I'll take a close look on the diff (y)

FreyJo, 2024-04-19:
I checked the commit in question and realized that I missed putting a breaking note in the release notes, which I just did.

> - Calling `ocp_nlp_precompute` is now required when creating an OCP solver from `C`, this is done in the C files generated by the template based interfaces (Python, Matlab and Octave). It might affect users that use an old / self written `C` file to set up their OCP solver. See: https://github.com/acados/acados/commit/37e17d31890ab54e5a855f1fe787fbf2f5d43bdb

However, I don't see how this can affect you given the workflow you described of regenerating the code with the latest release :thinking: 
Let me know if you find anything.
You could also try to bisect through the commits in the corresponding PR branch https://github.com/acados/acados/pull/888

Best,
Jonathan

closed 2025-04-08 by FreyJo