acados/acados #1111: Error during make install on MacOS (Intel CPU)

alexhansson, 2024-05-16:
Hello everyone,

I'm currently trying to install acados on my MacBook Pro with an Intel CPU. Everything including Cmake works, but there is an error when I try to `make install`, as outlined in the installation instructions. The following error appears:   

`error: call to undeclared library function 'malloc' with type 'void *(unsigned long)'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration].
void *raw_memory_ptr = malloc(memory_size);`

I managed to work around the issue by inserting `#include <stdlib.h>` in `external/qpoases/include/qpOASES_e/MessageHandling.h`. However, I'm wondering why this error appears and how I can resolve it. I think it might have to do with the underlying C version, but I am not entirely sure.

Thanks for the help.

FreyJo, 2024-05-17:
Hi,

this issue seems specific to the qpOASES version in acados, although you did not include the line where this occurs.
Actually, the acados qpOASES interface does not use the functions from qpOASES which use malloc inside.
Thus, we can either add the include as you suggested or comment/remove the functions that use malloc in the acados qpOASES version.
For now, I would recommend to keep the include you added or just compile `cmake -DACADOS_WITH_QPOASES=OFF ..`.

alexhansson, 2024-05-20:
Hi @FreyJo 

Thank you very much for answering.  
I uploaded a screenshot of the occurrence (see below).
Is there any way to fix this "cleanly" from my side, especially when thinking about docker-izing the project I'm working on with acados?
[image]

closed 2024-05-21 by sandmaennchen