acados/acados #1113: Error compiling Cython

kimanton, 2024-05-20:
I cloned and built the latest code
Then I tried to run Python example `crane/time_optimal_example.py`, however it fails to build solver
I don't know how to approach the errors it gives me. 
Any advice?

Here is terminal output. Does it have to do anything with python interface `acados_template`?
```
home/risc/acados1/include/../interfaces/acados_template/acados_template/acados_ocp_solver_pyx.pyx \
-I /home/risc/acados1/examples/acados_python/crane/c_generated_code \

Error compiling Cython file:
------------------------------------------------------------
...
            raise Exception(f"AcadosOcpSolver.eval_and_get_optimal_value_gradient(): Unknown field: with_respect_to = {with_respect_to}")
        return grad

    def eval_solution_sensitivity(self, stages: Union[int, List[int]], with_respect_to: str) \
                                               ^
------------------------------------------------------------

/home/risc/acados1/interfaces/acados_template/acados_template/acados_ocp_solver_pyx.pyx:299:48: undeclared name not builtin: Union

Error compiling Cython file:
------------------------------------------------------------
...
            raise Exception(f"AcadosOcpSolver.eval_and_get_optimal_value_gradient(): Unknown field: with_respect_to = {with_respect_to}")
        return grad

    def eval_solution_sensitivity(self, stages: Union[int, List[int]], with_respect_to: str) \
                                                          ^
------------------------------------------------------------

/home/risc/acados1/interfaces/acados_template/acados_template/acados_ocp_solver_pyx.pyx:299:59: undeclared name not builtin: List

Error compiling Cython file:
------------------------------------------------------------
...
        return grad

    def eval_solution_sensitivity(self, stages: Union[int, List[int]], with_respect_to: str) \
                -> Tuple[Union[List[np.ndarray], np.ndarray], Union[List[np.ndarray], np.ndarray]]:
                  ^
------------------------------------------------------------

/home/risc/acados1/interfaces/acados_template/acados_template/acados_ocp_solver_pyx.pyx:300:19: undeclared name not builtin: Tuple
make: *** [Makefile:151: ocp_cython_c] Error 1
Traceback (most recent call last):
  File "time_optimal_example.py", line 231, in <module>
    main(use_cython=use_cython)
  File "time_optimal_example.py", line 161, in main
    ocp_solver, integrator = setup_solver_and_integrator(x0, xf, N, use_cython)
  File "time_optimal_example.py", line 132, in setup_solver_and_integrator
    ocp_solver = AcadosOcpSolver.create_cython_solver('acados_ocp.json')
  File "/home/risc/acados1/interfaces/acados_template/acados_template/acados_ocp_solver.py", line 177, in create_cython_solver
    acados_ocp_solver_pyx = importlib.import_module(f'{os.path.split(code_export_directory)[1]}.acados_ocp_solver_pyx')
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 973, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'c_generated_code.acados_ocp_solver_pyx'

```

I didn't have such issue with a much older version of acados, however I don't remember which version it was

FreyJo, 2024-05-21:
Hi,

it seems to be related to the type hints in the cython wrapper.
What Python version are you using?
Can you try to upgrade?

kimanton, 2024-05-21:
Hi,
I am on Ubuntu 20.04, using Python 3.8.10
Unfortunately, I can't upgrade for now because I am using ROS1

FreyJo, 2024-05-21:
yes

FreyJo, 2024-05-21:
> Then, I rebuilt acados, removed and installed acados_template

And that fixed the issue?
I guess you need to recompile the acados core shared library from what the error says.

kimanton, 2024-05-21:
I deleted `build`, `include` and `lib` folders from acados `root_dir`. 
Then I recompiled acados, exported `LD_LIBRARY_PATH` and `ACADOS_SOURCE_DIR`, and installed `acados_template`

However I still get the same error
```
Traceback (most recent call last):
  File "time_optimal_example.py", line 231, in <module>
    main(use_cython=use_cython)
  File "time_optimal_example.py", line 161, in main
    ocp_solver, integrator = setup_solver_and_integrator(x0, xf, N, use_cython)
  File "time_optimal_example.py", line 132, in setup_solver_and_integrator
    ocp_solver = AcadosOcpSolver.create_cython_solver('acados_ocp.json')
  File "/home/risc/acados1/interfaces/acados_template/acados_template/acados_ocp_solver.py", line 177, in create_cython_solver
    acados_ocp_solver_pyx = importlib.import_module(f'{os.path.split(code_export_directory)[1]}.acados_ocp_solver_pyx')
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 657, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 556, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1166, in create_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
ImportError: /home/risc/acados1/examples/acados_python/crane/c_generated_code/acados_ocp_solver_pyx.so: undefined symbol: ocp_nlp_eval_lagrange_grad_p
```

FreyJo, 2024-05-21:
From your path `/home/risc/acados1`, I guess you have another acados installation lying around which it is trying to link to.
Please make sure to set the environment variables correctly corresponding to the installation you want to use.

kimanton, 2024-05-21:
Yes, I exported the correct installation like the following
```
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:"/home/risc/acados1/lib"
export ACADOS_SOURCE_DIR="/home/risc/acados1"
```
And I added it to `.bashrc` to make sure I always export the correct one

closed 2024-10-29 by FreyJo