acados/acados #1114: Batch OCP Solver

itaouil, 2024-05-20:
Hi,

I recently started to use the AcadoBatchSolver in order to solve multiple OCPs in parallel using OpenMP.

The solving part works great, but when I need to populate several OCPs with different constraints and states, it takes a lot of time just to set the OCPs. Do you have any recommendations on how to speed up this part (maybe there is some ACADOS feature that I could use)?

Best,
Ilyass

itaouil, 2024-11-10:
Hi @FreyJo,

I would be happy to try and contribute to this feature. Can you share some references or directions on how I could start with this feature?

Best,
Ilyass

sandmaennchen, 2024-11-12:
Hi Ilyass,

two recent features that might speed up setting up the OCPs:

- global parameters: If your parameters are not changing across stages, you can use `p_global` instead of `p`. Global parameters also make use of precomputation , cf. `AcadosOcpSolver.set_p_global_and_precompute_dependencies()`.
 (This requires CasADi pre-relase, cf. https://github.com/acados/acados/pull/1234)
- flat setters and getters: `AcadosOcpSolver.set_flat()` and `AcadosOcpSolver.get_flat()` allow to set/get the iterate for the whole OCP. This is much faster than setting/getting stagewise.

Regarding the batch solver: I just added a batched version of `set_flat` to the `AcadosOcpBatchSolver` in this PR https://github.com/acados/acados/pull/1349. The changes should be rather minimal and might be a good start for you to further extend the batch solver.  Feel free to reach out if something's unclear! 

Best, Katrin

sandmaennchen, 2024-11-18:
More functionality has been implemented in https://github.com/acados/acados/pull/1360.

FreyJo, 2025-02-10:
This has become a very "generic" feature request. If more specific functionality is needed, please open a new issue.

closed 2025-02-10 by FreyJo