acados/acados #1124: Cmake find_package problem when using acados with daqp

FranekStark, 2024-05-31:
When installing acados with daqp, and trying to find acados in another CMAKE project using `find_package(acados REQUIRED)`, the acadosConfig fails to find daqp:
```
CMake Error at CMakeLists.txt:43 (find_package):
  Found package configuration file:

    /opt/acados/cmake/acadosConfig.cmake

  but it set acados_FOUND to FALSE so package "acados" is considered to be
  NOT FOUND.  Reason given by package:

  The following imported targets are referenced, but are missing: daqp
```

The issue seems to be that daqp provides a `daqp.cmake` instead of a `daqpConfig.cmake` file.
The following command solves the issue for me:
```sudo ln -s /opt/acados/cmake/daqp.cmake /opt/acados/cmake/daqpConfig.cmake ```

FreyJo, 2024-06-17:
I think this might be fixed within DAQP, maybe @darnstrom knows

darnstrom, 2024-06-17:
Thanks for reporting this! I think it is just a matter of changing the line
`install(EXPORT ${PROJECT_NAME} DESTINATION cmake)`
to 
`install(EXPORT ${PROJECT_NAME}Config DESTINATION cmake)` 
in DAQP's CMakeLists.txt. 

I will have a look into it and fix it in future versions. I might also take the opportunity then to make acados use the latest version of DAQP (0.5.0 rather than 0.4.0)

closed 2024-11-18 by FreyJo