acados/acados #1125: Error compiling Acados with GCC 14.1.1: `implicit declaration of function ‘posix_memalign’`

hshose, 2024-05-31:
Compiling acados with up-to-date `gcc 14.1.1 20240522`, I get the following error:

```
acados/external/blasfeo/auxiliary/blasfeo_stdlib.c: In function ‘blasfeo_malloc_align’:
acados/external/blasfeo/auxiliary/blasfeo_stdlib.c:72:19: error: implicit declaration of function ‘posix_memalign’ [-Wimplicit-function-declaration]
   72 |         int err = posix_memalign( ptr, CACHE_LINE_SIZE, size );
      |                   ^~~~~~~~~~~~~~
make[2]: *** [external/blasfeo/CMakeFiles/blasfeo.dir/build.make:93: external/blasfeo/CMakeFiles/blasfeo.dir/auxiliary/blasfeo_stdlib.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
```

Downgrading to `gcc 13.2.1-6 20240417` resolves the issue for now.

hshose, 2024-05-31:
Proposed fix in [blasfeo PR](https://github.com/giaf/blasfeo/pull/175)

tmmsartor, 2024-06-22:
Related [commend](https://github.com/giaf/blasfeo/pull/175#issuecomment-2184079588)
Changing compilation standard for GNU/Clang case from `-std=c99` to `-std=gnu99` can also fix the issue.
If `acados` is not using gnu specific C non-standard extensions using `-std=gnu99` should not results in any differences, but that must be checked.
Apparently one difference is that some warnings are considered errors, namely `incompatible-pointer-types` and `int-conversion` but this can be overridden with `-Wno-incompatible-pointer-types` and `-Wno-int-conversion`.

NB: to see the different C-macro defined in the two cases one can run something like:
```bash
touch empty.c
gcc -std=gnu99 -dM -E empty.c > gnu99.c
gcc -std=c99 -dM -E empty.c > c99.c
vimdiff gnu99.c c99.c
```

tmmsartor, 2024-08-13:
Another way to fix this is to set to pass that flag from `acados` i.e. in the way of #1194 .

Also worth to mention that when trying to compile blasfeo without adding additional C_FLAGS like `c99` (which also diasable  the posix_memalign  definition) unfolds to ramification which ultimately make same test to fails, even if it does not seem a catastrophic failure (`Exception: acados solver took 23 iterations, expected range(17, 23).`)

As a related but separate topic, I am wondering if there are any drawback in adding more modularity with acados dependencies, i.e. linking to libraries instead of cloning submodules.

tmmsartor, 2024-08-16:
Just to summarize #1194 an easy workaround (for conventional platforms) would be adding: `-DCMAKE_C_FLAGS="-D_POSIX_C_SOURCE=200112L"` when calling `cmake` in `acados`.

giaf, 2024-11-05:
With this commit https://github.com/giaf/blasfeo/commit/01705e58f3cf88138749702e0f72831efb198be8 any dependency on posix_memalign should be gone.

closed 2024-11-06 by FreyJo