acados/acados #1130: cmakelists does not respect CMAKE_INSTALL_PREFIX

asherikov, 2024-06-07:
One of CMAKE_INSTALL_PREFIX use cases is to provide it via command line or a toolchain, which is broken by https://github.com/acados/acados/blob/master/CMakeLists.txt#L84C5-L84C25. Using a custom parameter ACADOS_INSTALL_DIR to override the default one requires package specific workarounds downstream.

One option is to use https://cmake.org/cmake/help/latest/variable/CMAKE_INSTALL_PREFIX_INITIALIZED_TO_DEFAULT.html instead, it would require bumping required cmake version to 3.7.1, which seems reasonable. If there are no objections I can do an MR.

FreyJo, 2024-06-17:
I think requiring CMake >= 3.7.1 is fine, apparently it was released in 2016.

asherikov, 2024-06-18:
Opened a PR -> https://github.com/acados/acados/pull/1138, closing this.