acados/acados #1131: Linearly dependent constraints at solution due to the initial state constraint formulation

trymte, 2024-06-09:
Appreciate the great work you have put into this framework. I have found it very useful.

I want to shed light on some problematic aspects with the formulation of the initial state constraints as two separate inequalities instead of a single equality constraint: 
[image]
```
def solve_for_x0(self, x0_bar, fail_on_nonzero_status=True, print_stats_on_failure=True):
        """
        Wrapper around `solve()` which sets initial state constraint, solves the OCP, and returns u0.
        """
        self.set(0, "lbx", x0_bar)
        self.set(0, "ubx", x0_bar)
```
At the optimal solution for any problem where `x(0) = x0_bar` is active, this will in acados lead to both `x(0) <= x0_bar` and `x0_bar <= x(0)` being active and linearly dependent inequality constraints. In turn, the linear independence constraint qualification (LICQ) condition for this problem formulation will not be satisfied. 

When extracting multipliers from the solver after a solution has been obtained, one therefore ideally want to extract only 1 set of multipliers for this initial state constraint. However, in the current version there will be multipliers for both `x(0) <= x0_bar` and `x0_bar <= x(0)`.

Correct me if I am wrong here, but if not, can the initial state constraint be handled as an equality constraint? This would make the problem formulation well-posed and satisfy LICQ when `x(0) = x0_bar`

sandmaennchen, 2024-06-11:
Hi :wave: 

in theory, you are right and LICQ is violated for the initial state constraint. In practice, this works fine and – depending on the QP solver you are using – the two inequalities might even get replaced by an equality.

Regarding sensitivities of the optimal solution with respect to the initial state, this function might be helpful for you:
https://github.com/acados/acados/blob/d7d91dc355433ac728bc1e23c8d8bd401e183903/interfaces/acados_template/acados_template/acados_ocp_solver.py#L499

trymte, 2024-06-11:
This problem only arose when using casadi functions on the side to compute sensitivities wrt. parameters, of solutions provided by Acados. It will be good to add this as a note in the docs for people who are using the (multipliers) from the solution externally.

I have considered using the sensitivity functions in acados, but they are unavailable when using regularization in the problem. Can they be extended to be compatible with some of the regularization methods available?  Also, does acados compute sensitivities w.r.t. parameters in the cost function, dynamics and also constraints?

sandmaennchen, 2024-06-11:
Sensitivities of the optimal value wrt. the initial state/params should work also with regularization. The solution sensitivities, i.e. $\partial x^* / \partial \bar{x}_0$,  $\partial u^* / \partial \bar{x}_0$, do not work for regularized problems since the sensitivities computed from a regularized Hessian might be incorrect.

Sensitivities with respect to parameters cover parametric external costs and parametric discrete dynamics at the moment and in addition assumes global parameters,  parameters need to be the same across all shooting nodes.

And you are of course right, documentation could be improved ...

sandmaennchen, 2024-06-19:
Some more notes on current status of solution sensitivities have been add in https://github.com/acados/acados/pull/1137.

closed 2024-06-19 by sandmaennchen

trymte, 2024-06-19:
Thanks. From a practical standpoint it would still be beneficial to use the sensitivities even though the problem is regularized. Then the user can decide whether to use the sensitivity results or not. Can this be made possible?

sandmaennchen, 2024-06-19:
You can allow this by commenting out the respective checks in the python interface:
https://github.com/acados/acados/blob/d7d91dc355433ac728bc1e23c8d8bd401e183903/interfaces/acados_template/acados_template/acados_ocp_solver.py#L582

But I think we won't allow this in general.

The recommended way is to have to solvers: One solver with regularized and/or Gauss-Newton Hessian for solving the OCP and then a second exact Hessian solver for evaluating solution sensitivities. The second solver can be initialized at the solution found by the first solver using `store_iterate` and `load_iterate`. This approach is implemented in the example [here](https://github.com/acados/acados/blob/master/examples/acados_python/pendulum_on_cart/solution_sensitivities/policy_gradient_example.py). 

Hope this is helpful!