acados/acados #1186: processing the command: make shared_library with the following error

ZhengCanlun, 2024-08-06:
external_function_generic.c: In function ‘external_function_external_param_casadi_assign’:
external_function_generic.c:1496:97: error: ‘external_function_external_param_casadi’ {aka ‘struct <anonymous>’} has no member named ‘np’
 1496 | emory + external_function_external_param_casadi_calculate_size(fun, fun->np) >=
      |                                                                        ^~

external_function_generic.c:1496:34: error: too many arguments to function ‘external_function_external_param_casadi_calculate_size’
 1496 |     assert((char *) raw_memory + external_function_external_param_casadi_calculate_size(fun, fun->np) >=
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
external_function_generic.c:1381:15: note: declared here
 1381 | acados_size_t external_function_external_param_casadi_calculate_size(external_function_external_param_casadi *fun)
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [<内置>：external_function_generic.o] 错误 1
make[2]: 离开目录“/home/iusl/acados/acados/utils”
make[1]: *** [Makefile:39：obj] 错误 2
make[1]: 离开目录“/home/iusl/acados/acados”
make: *** [Makefile:169：shared_library] 错误 2

sandmaennchen, 2024-08-07:
Hi :wave: 
this seems related to the recent changes to the internal handling of parameters, https://github.com/acados/acados/pull/1163. Please make sure that (1) you have the most recent acados version, (2) this version is properly installed, and (3) the solver code for your example is re-generated.

ZhengCanlun, 2024-08-07:
Thank you for your answer. We have been following the instructions step by step. The problem occurs at the 'make shared_library' step. We have not been able to resolve this issue so far. However, we were able to install it using a very older version.

sandmaennchen, 2024-08-07:
Could you provide the acados version/commit, as well as your operating system? Which python/Matlab version are you using?
Did you delete the folder `c_generated_code` created during solver creation before trying again?

sandmaennchen, 2024-08-08:
And did you compile with `BUILD_TYPE` set to debug?

sandmaennchen, 2024-08-09:
The issue should be fixed. Could you check whether you can successfully install acados using the current master? Thanks a lot!

closed 2024-08-16 by sandmaennchen