acados/acados #1224: make example_sim in c_generated_code for discrete time dynamics

th1991-01, 2024-08-28:
I run `acados/examples/acados_python/generic_dyn_disc/main.py` to generate C code. In `acados/examples/acados_python/generic_dyn_disc/c_generated_code/`, `make example` works fine, but `make example_sim` fails with the following error message,
```
acados_sim_solver_lin_mass.c: In function ‘lin_mass_acados_sim_create’:
acados_sim_solver_lin_mass.c:83:23: error: ‘DISCRETE’ undeclared (first use in this function)
   83 |     plan.sim_solver = DISCRETE;
      |                       ^~~~~~~~
acados_sim_solver_lin_mass.c:83:23: note: each undeclared identifier is reported only once for each function it appears in
make: *** [<ビルトイン>: acados_sim_solver_lin_mass.o] エラー 1

```

I also try the same thing for `acados/examples/acados_python/generic_impl_dyn/minimal_example_ocp_generic_impl_dyn.py`. In that case, both `make example` and `make example_sim` work fine. 

It seems that `IRK` is defined in C but `DISCRETE` is not.

sandmaennchen, 2024-08-28:
Hey :wave: 

acados does not provide a standalone integrator for discrete dynamics, you can directly use the provided CasADi function/generic function. But you should of course get a proper error messages instead of the compile error above.

th1991-01, 2024-08-28:
Thank you for your reply!

> acados does not provide a standalone integrator for discrete dynamics, you can directly use the provided CasADi function/generic function.

Does it mean the following?
* `make example_sim` provides an example of standalone integrator.
* In case of discrete time dynamics, as the provided CasADi function/generic function can be used directly, its standalone integrator example is not needed.

If so, I can understand the error. However, in that case, `main_sim_lin_mass.c` generated by `acados/examples/acados_python/generic_dyn_disc/main.py` might be a little confusing, as it might look like a standalone example.

sandmaennchen, 2024-08-28:
Yes exactly, we will add an additional check in order to not generate `main_sim_lin_mass.c` if discrete dynamics are used.

closed 2024-08-29 by FreyJo