acados/acados #1237: libblasfeo.so: undefined symbol: kernel_dpack_buffer_ft

giulioturrisi, 2024-09-03:
Hi, 

I recently installed acados from zero in two different PC, and I have noticed the following error related with blasfeo when launching my controller.

libblasfeo.so: undefined symbol: kernel_dpack_buffer_ft

The issue both appear in ubuntu 22/24, using different version of python, both with the latest acados master and older acados commit.

Giulio

FreyJo, 2024-09-03:
Hi Giulio,
could you provide some more information:
How do you compile acados?
Do you have a minimal example to reproduce this?
What CPU are you using?
This would greatly help to look into it.
Best,
Jonathan

giulioturrisi, 2024-09-03:
Sure! So, the instruction and example are here:
https://github.com/iit-DLSLab/Quadruped-PyMPC

The CPU are an Intel i7-13700h and and i7-10gen (on the two different machines i tested It)

giulioturrisi, 2024-09-03:
As additional information, it was casadi. In my yaml file, i was installing its lates version (along with pinocchio https://github.com/stack-of-tasks/pinocchio), which create a libblasfeo.so that is incompatible with acados. 
Setting as requirement ==3.6.5 (the latest supported version by acados), solved this problem since no libblasfeo.so is created.

closed 2024-09-03 by giulioturrisi