acados/acados #1259: Only generate and use Hessian functions needed when options exact_hess_constr, exact_hess_cost, exact_hess_dyn are off

FreyJo, 2024-09-25:
In case of `p_global` this might lead to errors otherwise.
Otherwise the additional functions only cause overhead when creating the solver and some additional memory requirements.

hrishikesh-arma, 2026-09-13:
A data point on the severity, in case it helps prioritise this: for `NONLINEAR_LS` + `GAUSS_NEWTON` the unused Hessians are not only solver-creation overhead — they can make the build impossible.

On a 24-joint manipulator whose residual runs through a tool transform with an extra rotary DOF, `generate_c_code_nls_cost` emits:

| generated file | size |
|---|---|
| `<model>_cost_y_0_hess.c` | 24.4 MB |
| `<model>_cost_y_hess.c` | 24.4 MB |
| `<model>_cost_y_e_hess.c` | 1.9 MB |

The generated `acados_solver_<model>.c` contains **zero** references to these symbols under `GAUSS_NEWTON`, but they are still in `OCP_SRC` in the Makefile, so they are compiled and linked.

Compiling the 24.4 MB unit at the default `ext_fun_compile_flags = '-O2'` drove `cc1` to **20.5 GB** resident, where it was killed by the kernel OOM killer. Because that is the *global* OOM killer, the process it selected was not the build — it killed the desktop session, so the failure did not even present as a build failure. For comparison, the largest *used* translation unit in the same build is the constraint Jacobian at 5.8 MB (~3.2 GB peak at `-O2`). The dead Hessians are by themselves the difference between a build that completes and one that cannot.

For anyone hitting this before the option lands, the workaround is to split generation from the build and drop the unreferenced entries:

```python
AcadosOcpSolver.generate(ocp, json_file=json_path)
# remove the 'OCP_SRC+= ..._hess.c' lines whose symbols do not appear
# in acados_solver_<model>.c
AcadosOcpSolver.build(code_export_dir, with_cython=False)
solver = AcadosOcpSolver(None, json_file=json_path, generate=False, build=False)
```

Checking each symbol against the generated solver source first keeps that safe against a future version that does wire the Hessians up.

One implementation note for whoever picks this up: in `generate_c_code_nls_cost` the expensive part runs before the emit —

```python
y_adj  = ca.jtimes(y_expr, ca.vertcat(u, x), y, True)
y_hess = ca.jacobian(y_adj, ca.vertcat(u, x), {"symmetric": is_casadi_SX(x)})
```

so guarding only the `add_function_definition` would still pay the CasADi `jacobian` of the adjoint. Both the construction and the emit want to be inside the condition. Note also that unlike `generate_c_code_discrete_dynamics`, `generate_c_code_explicit_ode`, `generate_c_code_implicit_ode` and `generate_c_code_constraint`, the NLS cost generator does not currently consult `opts.generate_hess` at all, so it is unguarded even for the plain `hessian_approx='GAUSS_NEWTON'` case rather than only the `exact_hess_*` sub-options.

Checked against master (`interfaces/acados_template/acados_template/casadi_function_generation.py`) as well as v0.5.4 — still unguarded in both.

Happy to open a PR for the NLS cost generator if that would be useful.

FreyJo, 2026-09-14:
Thanks for the comment.
I fixed your specific issue here: https://github.com/acados/acados/pull/1924