acados/acados #1306: Failed to compile (matlab Windows) generated c code in ubuntu

ShengpingY, 2024-10-20:
Hi,

I am currently new to acados,

I have tried the pendulum example from .\acados\examples\acados_matlab_octave\getting_started and get the c_generated_code successfully, then I tried to transplant the generated code to ubuntu by modification of 'ACADOS_INCLUDE_PATH' and 'ACADOS_LIB_PATH' in CMakeList.txt into path of acados in ubuntu
```
# set some search paths for preprocessor and linker
set(ACADOS_INCLUDE_PATH /home/shengping/acados/examples/acados_matlab_octave/../../include CACHE PATH "Define the path which contains the include directory for acados.")
set(ACADOS_LIB_PATH /home/shengping/acados/examples/acados_matlab_octave/../../lib CACHE PATH "Define the path which contains the lib directory for acados.")
```
and change option of 'BUILD EXAMPLE' from 'OFF' to 'ON' 
`option(BUILD_EXAMPLE "Should the example main_pendulum be build?" ON)`

However when trying to compile the code with CMake with the following command:
```
makedir -p build
cd build
cmake ..
make -j1
```
I got a lot of errors reminding too few arguments to some generated solver, with the following log:
```
[  5%] Building C object CMakeFiles/ocp_pendulum.dir/pendulum_cost/pendulum_cost_y_0_fun.c.o
[ 11%] Building C object CMakeFiles/ocp_pendulum.dir/pendulum_cost/pendulum_cost_y_0_fun_jac_ut_xt.c.o
[ 17%] Building C object CMakeFiles/ocp_pendulum.dir/pendulum_cost/pendulum_cost_y_fun.c.o
[ 23%] Building C object CMakeFiles/ocp_pendulum.dir/pendulum_cost/pendulum_cost_y_fun_jac_ut_xt.c.o
[ 29%] Building C object CMakeFiles/ocp_pendulum.dir/pendulum_cost/pendulum_cost_y_e_fun.c.o
[ 35%] Building C object CMakeFiles/ocp_pendulum.dir/pendulum_cost/pendulum_cost_y_e_fun_jac_ut_xt.c.o
[ 41%] Building C object CMakeFiles/ocp_pendulum.dir/pendulum_constraints/pendulum_constr_h_0_fun.c.o
[ 47%] Building C object CMakeFiles/ocp_pendulum.dir/pendulum_constraints/pendulum_constr_h_0_fun_jac_uxt_zt.c.o
[ 52%] Building C object CMakeFiles/ocp_pendulum.dir/pendulum_constraints/pendulum_constr_h_fun.c.o
[ 58%] Building C object CMakeFiles/ocp_pendulum.dir/pendulum_constraints/pendulum_constr_h_fun_jac_uxt_zt.c.o
[ 64%] Building C object CMakeFiles/ocp_pendulum.dir/acados_solver_pendulum.c.o
/home/shengping/acados/examples/acados_matlab_octave/c_generated_code/acados_solver_pendulum.c: In function ‘pendulum_acados_create_setup_functions’:
/home/shengping/acados/examples/acados_matlab_octave/c_generated_code/acados_solver_pendulum.c:329:9: error: too few arguments to function ‘external_function_external_param_casadi_create’
         external_function_external_param_casadi_create(&capsule->__CAPSULE_FNC__ ); \
         ^
/home/shengping/acados/examples/acados_matlab_octave/c_generated_code/acados_solver_pendulum.c:329:9: note: in definition of macro ‘MAP_CASADI_FNC’
         external_function_external_param_casadi_create(&capsule->__CAPSULE_FNC__ ); \
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/shengping/acados/examples/acados_matlab_octave/c_generated_code/acados_solver_pendulum.c:38:0:
/home/shengping/acados/examples/acados_matlab_octave/../../include/acados_c/external_function_interface.h:90:6: note: declared here
 void external_function_external_param_casadi_create(external_function_external_param_casadi *fun, external_function_opts *opts_);
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/shengping/acados/examples/acados_matlab_octave/c_generated_code/acados_solver_pendulum.c:329:9: error: too few arguments to function ‘external_function_external_param_casadi_create’
         external_function_external_param_casadi_create(&capsule->__CAPSULE_FNC__ ); \
         ^
/home/shengping/acados/examples/acados_matlab_octave/c_generated_code/acados_solver_pendulum.c:329:9: note: in definition of macro ‘MAP_CASADI_FNC’
         external_function_external_param_casadi_create(&capsule->__CAPSULE_FNC__ ); \
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/shengping/acados/examples/acados_matlab_octave/c_generated_code/acados_solver_pendulum.c:38:0:
/home/shengping/acados/examples/acados_matlab_octave/../../include/acados_c/external_function_interface.h:90:6: note: declared here
 void external_function_external_param_casadi_create(external_function_external_param_casadi *fun, external_function_opts *opts_);
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/shengping/acados/examples/acados_matlab_octave/c_generated_code/acados_solver_pendulum.c:329:9: error: too few arguments to function ‘external_function_external_param_casadi_create’
         external_function_external_param_casadi_create(&capsule->__CAPSULE_FNC__ ); \
         ^
/home/shengping/acados/examples/acados_matlab_octave/c_generated_code/acados_solver_pendulum.c:329:9: note: in definition of macro ‘MAP_CASADI_FNC’
         external_function_external_param_casadi_create(&capsule->__CAPSULE_FNC__ ); \
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/shengping/acados/examples/acados_matlab_octave/c_generated_code/acados_solver_pendulum.c:38:0:
/home/shengping/acados/examples/acados_matlab_octave/../../include/acados_c/external_function_interface.h:90:6: note: declared here
 void external_function_external_param_casadi_create(external_function_external_param_casadi *fun, external_function_opts *opts_);
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/shengping/acados/examples/acados_matlab_octave/c_generated_code/acados_solver_pendulum.c:329:9: error: too few arguments to function ‘external_function_external_param_casadi_create’
         external_function_external_param_casadi_create(&capsule->__CAPSULE_FNC__ ); \
         ^
/home/shengping/acados/examples/acados_matlab_octave/c_generated_code/acados_solver_pendulum.c:329:9: note: in definition of macro ‘MAP_CASADI_FNC’
         external_function_external_param_casadi_create(&capsule->__CAPSULE_FNC__ ); \
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... 113 lines cut ...]
make[1]: *** [CMakeFiles/ocp_pendulum.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

```

Does someone know where this comes from? Or how can I implement the generated C-code from Matlab into my project?

By the way, I have also tested my own model and constraint and tried with a for-loop to continuously update the initial value and reference value and continuously solving the problem, however when checking the generated c-code, I did not find a part which updates the value with for-loop, should I implement the updating separately in C or is there anything I missed?
```
for rr = 1:100
    tic;
    % updating initial state
    x0 = update_x0(x0,[-0.5,0.5]);
    %updating reference value
    ocp.cost.yref_0 = generate_random_matrix(ny_0, 1, [-0.5,0.5]);
    ocp.cost.yref = generate_random_matrix(ny, 1, [-0.5,0.5]);
    ocp.cost.yref_e = generate_random_matrix(ny_e, 1, [-0.5,0.5]);
    % ocp_solver = AcadosOcpSolver(ocp);
    ocp_solver.solve();
    toc
end
```

Thanks!

FreyJo, 2024-10-21:
Hi,
you should make sure that the acados installations on both machines match.
In acados, the generated code is adapted when changes in the C part of acados are introduced.
Thus, generated acados code is tested with that specific acados version but not guarenteed to work with other acados versions.
The issues you are seeing correspond to changes in the C part of acados in #1163 and #1296 
Let me know if that allows you to fix the issue!

ShengpingY, 2024-10-21:
> Hi, you should make sure that the acados installations on both machines match. In acados, the generated code is adapted when changes in the C part of acados are introduced. Thus, generated acados [...]

Hi,

Thanks a lot!

The problem was caused by a different version of Acados, after I changed both versions to v0.4.0 the problem was gone.

closed 2024-10-21 by ShengpingY