acados/acados #1318: A compile error when run “make examples_c”

liu2050, 2024-10-28:
I try to build and install acados in ubuntu 20.04, it all goes well until I run the “make examples_c”, there is an make error:
```
acados/lib/libacados.so: undefined reference to `ocp_nlp_globalization_merit_backtracking_config_initialize_default'
acados/lib/libacados.so: undefined reference to `ocp_nlp_globalization_config_calculate_size'
acados/lib/libacados.so: undefined reference to `ocp_nlp_ddp_config_initialize_default'
acados/lib/libacados.so: undefined reference to `ocp_nlp_globalization_config_assign'
acados/lib/libacados.so: undefined reference to `ocp_nlp_globalization_funnel_config_initialize_default'
acados/lib/libacados.so: undefined reference to `ocp_nlp_globalization_merit_backtracking_ddp_needs_qp_objective_value'
acados/lib/libacados.so: undefined reference to `ocp_nlp_globalization_fixed_step_config_initialize_default'
acados/lib/libacados.so: undefined reference to `ocp_nlp_globalization_merit_backtracking_find_acceptable_iterate_for_ddp'
```
How to solve the problem? Thank you.

liu2050, 2024-10-28:
[image]

FreyJo, 2024-10-28:
Hi,
as stated in the documentation, CMake is our preferred build system and the one that is actively tested on Github actions.
In https://github.com/acados/acados/pull/1319, I added more notes stating that CMake should be used and Make is not tested and fixed the error you reported here.

closed 2024-10-28 by sandmaennchen