acados/acados #1328: How to transform libacados.so from x86-64 to arm?

liu2050, 2024-10-29:
I want to use acados in stm32, so I have to use arm_none_eabi_gcc. But when I compile the project, the following errors occurred.
```
acados/lib/libacados.so: file not recognized: file format not recognized
```
I guess that libacados.so needs to be changed into arm.
How to solve it? Thank you.

FreyJo, 2024-10-29:
You need to cross compile acados.
You need to do something similar to step 1-2 described here:
https://docs.acados.org/embedded_workflow/index.html
i.e 1) CMake toolchain file for your system 2) Cross-compile acados for your platform.
I hope you manage to do so!
It would be great if you could contribute your toolchain file for other users once you managed to do so.

josipkh, 2024-10-29:
Maybe the following links will also be helpful:
[STM32H7 example (forum)](https://discourse.acados.org/t/working-example-for-stm32h7-cortex-m7-32-bit/406)
[STM32H7 example (GitHub)](https://github.com/mindThomas/acados-STM32)
[acados on embedded platform (forum)](https://discourse.acados.org/t/acados-on-embedded-platform/78)

liu2050, 2024-10-29:
Thank you both for your detailed explanations. With your help, the issue has been resolved. Here is the CMake toolchain file I wrote:
'''
# Set the CMake system name to Generic, as we are cross-compiling for ARM
SET(CMAKE_SYSTEM_NAME Generic)

# Set the system processor to ARM
SET(CMAKE_SYSTEM_PROCESSOR arm)

# Set the toolchain to use the arm-none-eabi toolchain
SET(TOOLCHAIN arm-none-eabi)

# Set the build type to static library by default
SET(CMAKE_BUILD_TYPE STATIC_LIBRARY)

# Set the prefixes and suffixes for shared and static libraries
SET(CMAKE_SHARED_LIBRARY_PREFIX "lib")
SET(CMAKE_SHARED_LIBRARY_SUFFIX ".so")
SET(CMAKE_STATIC_LIBRARY_PREFIX "lib")
SET(CMAKE_STATIC_LIBRARY_SUFFIX ".a")

# Find the arm-none-eabi-gcc compiler
FIND_PROGRAM(ARM_NONE_EABI_GCC NAMES arm-none-eabi-gcc PATHS /usr/bin/ NO_DEFAULT_PATH)

# Check if the compiler was found
IF(NOT ARM_NONE_EABI_GCC)
  MESSAGE(FATAL_ERROR "arm-none-eabi-gcc not found. Please set the ARM_NONE_EABI environment variable to the path where arm-none-eabi-gcc is located.")
ENDIF()

# Set the compiler paths
SET(CMAKE_C_COMPILER ${ARM_NONE_EABI_GCC})
SET(CMAKE_CXX_COMPILER ${TOOLCHAIN}-g++)

# Set the flags for the compiler
SET(CMAKE_C_FLAGS_DEBUG "-g")
SET(CMAKE_C_FLAGS_MINSIZEREL "-Os -DNDEBUG")
SET(CMAKE_C_FLAGS_RELEASE "-O3 -DNDEBUG")
SET(CMAKE_C_FLAGS_RELWITHDEBINFO "-O2 -g")

SET(CMAKE_CXX_FLAGS_DEBUG "-g")
SET(CMAKE_CXX_FLAGS_MINSIZEREL "-Os -DNDEBUG")
SET(CMAKE_CXX_FLAGS_RELEASE "-O3 -DNDEBUG")
SET(CMAKE_CXX_FLAGS_RELWITHDEBINFO "-O2 -g")

add_link_options("--specs=nosys.specs")

# Set the find root path to the target directory
SET(CMAKE_FIND_ROOT_PATH $ENV{ARM_NONE_EABI})

# Set the find root path mode to only search in the target directories
SET(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
SET(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
SET(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)

# Additional flags can be set here as needed
# acados flags
set(BLASFEO_TARGET "GENERIC" CACHE STRING "BLASFEO Target architecture")
set(HPIPM_TARGET "GENERIC" CACHE STRING "HPIPM Target architecture")
set(BUILD_SHARED_LIBS OFF CACHE STRING "Build shared libraries")
set(BLASFEO_EXAMPLES OFF CACHE BOOL "Examples disabled")
set(EXT_DEP OFF CACHE BOOL "Compile external dependencies in BLASFEO")
set(ACADOS_INSTALL_DIR "install" CACHE PATH  "Installation path to PROJECT_SOURCE_DIR")
set(USE_C99_MATH ON CACHE BOOL "Use C99 extension to math library")
```

liu2050, 2024-10-29:
Thank you both for your detailed explanations. With your help, the issue has been resolved. Here is the CMake toolchain file I wrote:
```
SET(CMAKE_SYSTEM_NAME Generic)

SET(CMAKE_SYSTEM_PROCESSOR arm)

SET(TOOLCHAIN arm-none-eabi)

SET(CMAKE_BUILD_TYPE STATIC_LIBRARY)

SET(CMAKE_SHARED_LIBRARY_PREFIX "lib")
SET(CMAKE_SHARED_LIBRARY_SUFFIX ".so")
SET(CMAKE_STATIC_LIBRARY_PREFIX "lib")
SET(CMAKE_STATIC_LIBRARY_SUFFIX ".a")

FIND_PROGRAM(ARM_NONE_EABI_GCC NAMES arm-none-eabi-gcc PATHS /usr/bin/ NO_DEFAULT_PATH)

SET(CMAKE_C_COMPILER ${ARM_NONE_EABI_GCC})
SET(CMAKE_CXX_COMPILER ${TOOLCHAIN}-g++)

SET(CMAKE_C_FLAGS_DEBUG "-g")
SET(CMAKE_C_FLAGS_MINSIZEREL "-Os -DNDEBUG")
SET(CMAKE_C_FLAGS_RELEASE "-O3 -DNDEBUG")
SET(CMAKE_C_FLAGS_RELWITHDEBINFO "-O2 -g")

SET(CMAKE_CXX_FLAGS_DEBUG "-g")
SET(CMAKE_CXX_FLAGS_MINSIZEREL "-Os -DNDEBUG")
SET(CMAKE_CXX_FLAGS_RELEASE "-O3 -DNDEBUG")
SET(CMAKE_CXX_FLAGS_RELWITHDEBINFO "-O2 -g")

add_link_options("--specs=nosys.specs")

SET(CMAKE_FIND_ROOT_PATH $ENV{ARM_NONE_EABI})

SET(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
SET(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
SET(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)

set(BLASFEO_TARGET "GENERIC" CACHE STRING "BLASFEO Target architecture")
set(HPIPM_TARGET "GENERIC" CACHE STRING "HPIPM Target architecture")
set(BUILD_SHARED_LIBS OFF CACHE STRING "Build shared libraries")
set(BLASFEO_EXAMPLES OFF CACHE BOOL "Examples disabled")
set(EXT_DEP OFF CACHE BOOL "Compile external dependencies in BLASFEO")
set(ACADOS_INSTALL_DIR "install" CACHE PATH  "Installation path to PROJECT_SOURCE_DIR")
set(USE_C99_MATH ON CACHE BOOL "Use C99 extension to math library")
```

closed 2024-10-30 by liu2050