acados/acados #1365: Nonsense integration and QP Bounds Violated

tchase24, 2024-11-20:
Branch: v0.3.5_stable
QP solvers: [PARTIAL_CONDENSING_HPIPM, FULL_CONDENSING_QPOASES]
Interfaces: C & Python

- Results of the solver reported prediction horizon state are vastly different from state integration for the prediction horizon given the optimized control output(s) along the horizon.
  - Ex: 
    - Inputs: 
      - Pred horizon of 20, 0.01s time slices 
      - First order system with state x, and x_dot = u
      - Upper bound of 20, lower bound of 1 on x
      - Upper bound of 10, lower bound of -10 on u
      - Initial state x0 = 9
    - Ouputs:
      - Solved optimal control (u_traj[0]) = 39
      - Solved next state output (x_traj[0]) = -52
      - Integrated (verified) state = 9.39
- Box constraints are being violated (see previous example) and the json dump shows that they do not match with the input.
  - ubx_00 is 11, lbx_00 is -8 (makes sense)
  - ubx_01 to ubx_20 is 19, lbx_01 to lbx_20 = 0 (suspect)

Any ideas:
- Why are constraints being violated?
- Why is integration is inconsistent with the solver?
- Why does the json output not match the solver?
- Am I missing something?

sandmaennchen, 2024-11-20:
Regarding your issue with the integration:
- which integrator are you using to verify the integration? An `AcadosSimSolver`? Is this integrator using the same settings?
- Note that the next state is `x_traj[1]` and not `x_traj[0]` which is the initial state

Regarding constraint violation:

- What is the return code of the solver?
- Did you check the output of `solver.print_statistics()`? Does the solver converge?

tchase24, 2024-11-20:
@sandmaennchen thank you very much for your quick response! You have helped me remedy some of the problem(s), although there is still some error.

For the integration:
- I am using the `AcadosSimSolver` as well as verifying integration by hand. The integrator is using the same settings. I have constructed it using the same `AcadosOcp` as is being used by the `AcadosOcpSolver`, as well as using an `AcadosSim` with the model and settings copied.
- Given that `x_traj[1]` is the next state as a response to `u_traj[0]`, the integration does work out between subsequent stages in the prediction horizon. 
- If `x_traj[0]` is supposed to be the initial state then that is something inconsistent: I initialize the ocp with `x0=9` and it returns a different `x_traj[0]` .

For the constraints
- I am getting return code 0 from the solver for both qp solvers. See the next point first, but should this be happening if the qp solver shows an error?
- I was hitting the max qp iterations and getting status 2. After upping the max iterations, the output of `solver.print_statistics()`  for `FULL_CONDENSING_QPOASES` is
```
iter    qp_stat qp_iter
0       0       0
1       0       273
```
and for `PARTIAL_CONDENSING_HPIPM` it is:
```
iter    qp_stat qp_iter
0       0       0
1       0       53
```
- It looks like the solver is converging now, and I am seeing the state/control bounds being kept.

I would say my concern with the bounds and integration are now resolved. The only issue I am seeing now is the initial condition for the prediction horizon is not matching what is being input. I would greatly appreciate any assistance you can offer on the matter!

sandmaennchen, 2024-11-21:
Yes, the initial state constraint needs to be set via `solver.constraints_set(0,"ubx",x0)` and `solver.constraints_set(0,"lbx",x0)` (or by directly using `solver.solve_for_x0(x0)`).

From the output that you provided, it looks like you are using `SQP_RTI`. Is that intended? I would recommend to start with `SQP` to get a fully converged solution.

Regarding the QP status, `qp_stat=3` for HPIPM indicates that `NaN` values were detected. Please check all of your nonlinear functions whether they might include something like a division by zero, logarithm of a negative number, etc.

tchase24, 2024-11-21:
@sandmaennchen I did intend to use `SQP_RTI` as the goal is to integrate with a real-time system. I do not have any nonlinear functions with division, logarithms, roots, or any other other asymptotic function. My cost function is a simple weighted Euclidean norm with a positive definite hessian, although I am using the `PROJECT` regularize method due to my model. 

When changing to `SQP`, the solver does more iterations but eventually runs into the same `qp_stat=3`. Am I right in thinking my problem is ill-conditioned? 

I am still concerned with the `lbx_00` and `ubx_00` terms of the qp json dump not matching, but the rounds it does converge `x_traj[0]` is within floating point error of `x0` so maybe its just my lack of familiarity with the solver.

closed 2024-11-26 by tchase24