acados/acados #1373: How to update the model dynamically?

liu2050, 2024-11-26:
In my project, some dynamic parameters of the model will change with the state variables. Do I have to redefine the model and redefine the solver for each calculation?

liu2050, 2024-11-26:
Or could it be that the dynamically changing parameters are also considered as state variables?

sandmaennchen, 2024-11-26:
Not sure if I understand correctly, but can't you just include the dependence of $p$ on $x$ in your dynamics, i.e. use $f(x, u) = \tilde{f}(x, u, p(x))$?

liu2050, 2024-11-27:
I‘m sorry I may not fully understand your explanation. In my project, p(x) is used to calculate f_expl_expr, and is a piecewise function with respect to x. Therefore, I need to compare x with numerical values to switch different values of p(x). However, when building the model according to the MATLAB code examples, x is defined as a sym, which cannot be compared with numerical values. I have tried to use p(x) as a new state variable, i.e., x_new = [x; p(x)], which can indirectly achieve the target, but it doesn't feel like the best solution. So, I'm wondering if there is a better way to solve this?

Additionally, when implementing the above method, I encountered a new problem. The calculation of the f_expl_expr might result in NaN values. I want to use isnan( ) to check and assign zero values, but isnan( ) can only check numerical values and cannot be applied to sym. How can I resolve this issue?

sandmaennchen, 2024-11-27:
Is $p(x)$ non-differentiable wrt. $x$? In this case, you do not want to include it in the dynamics directly. If it does not change a lot with $x$, you might try to approximate the dynamics by using $f(x, u, p)$, i.e. omitting the dependence of the parameters on $x$ and set the parameters $p$ for each OCP/update $p$ manually after each SQP iteration.

liu2050, 2024-11-27:
I understand now, thank you for your patience in explaining.

closed 2024-11-27 by liu2050