acados/acados #1378: How can I use the isnan() function in model.m?

liu2050, 2024-11-28:
In model.m, when calculating xdot, inevitable NAN values sometimes occur, which leads to the failure of the solver's iteration. When using other NMPC libraries, the isnan() function is used for judgment; if it is NAN, it is assigned a value of 0. However, in model.m, the result of the formula will include sym, so isnan() cannot be used. I would like to ask how to use the isnan() function in model.m? Or are there any other ways to perform the aforementioned judgment and assignment?

sandmaennchen, 2024-11-28:
Hi, I think this is not possible as CasADi does not provide a `isnan()` check. Please check your model equations for operations that might result in nans and adapt them, e.g. if your model includes a term like $1/x^2$, you can approximate it with $1/(x^2 + \epsilon)$ for some small value $\epsilon > 0$ to avoid a division by zero.

liu2050, 2024-11-29:
Thank you very much for your answer; it has been extremely helpful to me. Thank you.

closed 2024-11-29 by liu2050