acados/acados #1389: setting lbx and ubx

goosst, 2024-12-01:
Hello,

This might not be a bug but didn't immediately see something obvious.
I'm getting this strange error where the lbx (and ubx) boundary should be set to dimension 0

```
-> ocp_solver.set(0, "lbx", fubar)
(Pdb) ocp_solver.set(0, "lbx", fubar)
*** Exception: AcadosOcpSolver.set(): mismatching dimension for field "lbx" with dimension 0 (you have 2)
(Pdb) print(fubar.shape) 
```

related to this section of the code, I've added my full code for completeness [ClosedLoop.zip](https://github.com/user-attachments/files/17969732/ClosedLoop.zip).
``` 
        pdb.set_trace()
        fubar=np.array([10, 10])
        ocp_solver.set(0, "lbx", fubar) #wtf
        print(fubar.shape) 
        ```
        

Although not related, I'm not getting any meaningful results out of the optimizer. I would have hoped a linear cost function would not limit the exploration of 2 inputs of my model. In casadi I did have something working with ipopt, but I need the c-code to run it elsewhere than a pc. So thought to quickly convert it to acados ...

FreyJo, 2024-12-01:
Hi,
you need to define the initial state constraint before defining your problem, using something like
`ocp.constraints.x0 = np.array([15,15])`
this line is commented in your code.
You can define more general constraints on `x` at the initial stage by setting
`ocp.constraints.lbx_0`, `ocp.constraints.ubx_0`, `ocp.constraints.idxbx_0`, please check the documentation of those fields.
The dimension of the problem is fixed when you create the solver and cannot be changed afterwards.

goosst, 2024-12-01:
lol :), thanks

closed 2024-12-01 by goosst