acados/acados #1450: Create constraints for x_0 by default

RudolfReiter, 2025-02-28:
Without specifying an initial state in the AcadosOcp object, no constraints are created for x_0. When setting them as part of the closed loop iterations, an error is thrown that the constraints have size zero.
Usually, the initial state is unknown when the AcadosOcp is created. Therefore, it should not be required to be set to set the initial state in order to simply create the constraints.

FreyJo, 2025-02-28:
In acados, the structure of the constraint and their dimensions are specified before creating the solver.
There are OCP structured problems without initial state constraint, such as MHE problems.
Additionally, just some components of the state could be constrained at node zero.
I agree that it is most common to have an equality constraint for `x0`, but it is not clear how such a default is to be implemented.
Note that we tried specifying the initial state constraint as easy as possible by wrapping the internals `lbx`, `ubx`, `idxbx` and `idxbxe`, as can be seen here https://github.com/acados/acados/blob/135343ad43285bbe34d7d26adfa5be39b62f2111/interfaces/acados_template/acados_template/acados_ocp_constraints.py#L831-L845

A clear suggestion on how to make this easier is of course welcome.

RudolfReiter, 2025-03-04:
Ok, the reasoning makes sense.

closed 2025-03-04 by RudolfReiter