acados/acados #1486: Linear and nonlinear solver yield different results with same parameters for a linear system

SimonRennotte, 2025-03-31:
Hi there,

First, thanks for taking the time developing and maintaining this library!

I started using it 2 weeks ago and I am still trying to understand how to use it effectively.

I currently trying to use it to control a Pan & tilt unit to track a moving object in a simulation (position of object on screen is assumed to be known)

When I use a nonlinear controller, it works fine but when I switch to a linear controller with the same parameters, it seems that it's as if the control signal u is too penalized. So the u value is lower, and the error on the tracking is thus higher.

I was wondering if this was due to a parameter I did not see that should be set differently for the linear mpc compared to the nonlinear mpc?
I assumed by keeping the Q and R matrix the same, that the regularization of the control signal would be the same in both cases?

For more details, I attached the results of both tests, and the code is below.
Any ideas ?

Thanks a lot for your time !

        ocp = AcadosOcp()
        ocp.model = self.model
        
        nx = self.model.x.rows()
        nu = self.model.u.rows()
        ocp.cost.cost_type = self.cost_type
        ocp.cost.cost_type_e = self.terminal_cost_type
        ocp.cost.yref = np.zeros((nx+nu,))
        ocp.cost.yref_e = np.zeros((nx,))
        ocp.cost.W = 2 * ca.diagcat(self.Q_mat, self.R_mat).full()
        ocp.cost.W_e = 2 * self.Q_mat * self.end_window_loss_coef

        ##### only difference is here #####
        if self.cost_type == "LINEAR_LS":
            ocp.cost.Vu = np.vstack((np.eye(nu), np.zeros((nx,nu))))
            ocp.cost.Vx = np.vstack((np.zeros((nu, nx)), np.eye(nx)))
            ocp.cost.Vx_e = np.eye(nx)
        else:
            # path cost
            ocp.model.cost_y_expr = ca.vertcat(self.model.x, self.model.u)
            # terminal cost
            ocp.model.cost_y_expr_e = self.model.x
        
        # set constraints
        ocp.constraints.lbu = np.array([-self.acc_max_pan, -self.acc_max_tilt])
        ocp.constraints.ubu = np.array([+self.acc_max_pan, +self.acc_max_tilt])
        ocp.constraints.idxbu = np.array([0, 1])
        ocp.constraints.x0 = np.array([obj_pos_x, obj_pos_y, 0, 0, 0, 0])
        
        ocp.constraints.lbx = np.array([-self.speed_max_pan, -self.speed_max_pan])
        ocp.constraints.ubx = np.array([+self.speed_max_tilt, +self.speed_max_tilt])
        ocp.constraints.idxbx = np.array([4, 5]) # v is x[2]
        # indices of slacked constraints within bx
        #ocp.constraints.idxsbx = np.array([0])
        
        # set options solver
        N = int(time_horizon / step_time)
        ocp.solver_options.N_horizon = N
        ocp.solver_options.tf = time_horizon
        ocp.solver_options.qp_solver = self.qp_solver
        ocp.solver_options.hessian_approx = self.hessian_approx 
        ocp.solver_options.integrator_type = self.integrator_type
        ocp.solver_options.print_level = self.print_level
        ocp.solver_options.nlp_solver_type = self.nlp_solver_type
        ocp.solver_options.globalization = self.globalization
        
        if not(os.path.exists(json_file)) or not(os.path.exists(os.path.join("c_generated_code", "libacados_ocp_solver_obj_tracking.so"))) or build:
            build = True
            generate = True
        else:
            build = False
            generate = False
        
        ocp_solver = AcadosOcpSolver(ocp, build=build, generate=generate, json_file=json_file)
        simX = np.zeros((N+1, nx))
        simU = np.zeros((N, nu))
        
        u0 = ocp_solver.solve_for_x0(x0_bar=np.array([obj_pos_x, obj_pos_y, obj_speed_x, obj_speed_y, pt_speed_pan, pt_speed_tilt]))

And here is the model:

          obj_pos_x = SX.sym('obj_pos_x')
          obj_pos_y = SX.sym('obj_pos_y')
          obj_speed_pan_state = SX.sym('obj_speed_pan')
          obj_speed_tilt_state = SX.sym('obj_speed_tilt')
          pt_speed_pan_state = SX.sym('pt_speed_pan')
          pt_speed_tilt_state = SX.sym('pt_speed_tilt')
      
          x = vertcat(obj_pos_x, obj_pos_y, obj_speed_pan_state, obj_speed_tilt_state, pt_speed_pan_state, pt_speed_tilt_state)
      
          pan_acc = SX.sym('pan_acc')
          tilt_acc = SX.sym('tilt_acc')
          u = vertcat(pan_acc, tilt_acc)
      
          # xdot
          obj_pos_x_dot = SX.sym('obj_pos_x_dot')
          obj_pos_y_dot = SX.sym('obj_pos_y_dot')
          obj_speed_pan_dot = SX.sym('obj_speed_pan_dot')
          obj_speed_tilt_dot = SX.sym('obj_speed_tilt_dot')
          pt_speed_pan_dot = SX.sym('pt_speed_pan_dot')
          pt_speed_tilt_dot = SX.sym('pt_speed_tilt_dot')
      
          xdot = vertcat(obj_pos_x_dot, obj_pos_y_dot, obj_speed_pan_dot, obj_speed_tilt_dot, pt_speed_pan_dot, pt_speed_tilt_dot)
      
          # dynamics
          f_expl = vertcat(x_res / x_fov * (obj_speed_pan_state - pt_speed_pan_state),
                           y_res / y_fov * (obj_speed_tilt_state - pt_speed_tilt_state), 
                           0, 0,
                           pan_acc, 
                           tilt_acc)

Nonlinear mpc results:
[image]

Linear mpc results:
[image]

sandmaennchen, 2025-03-31:
I think you need to switch the order of x and u when defining the linear least-squares cost. In the linear formulation you have (u, x) and in the nonlinear one (x, u) 🙃

SimonRennotte, 2025-04-03:
Yes, that was the problem, oops.
Thanks for the help!

closed 2025-04-03 by SimonRennotte