acados/acados #1517: Segfault on Windows in C example `examples/c/wind_turbine_nmpc.c` during development

FreyJo, 2025-05-02:
I had a mysterious Segfault only happening on Windows with `examples/c/wind_turbine_nmpc.c` while working on https://github.com/acados/acados/pull/1512
The issue only occurred when reusing the external function workspace in the example.
A branch to reproduce the issue can be found here: https://github.com/acados/acados/compare/main...FreyJo:acados:external_fun_reuse_windows_segfault_issue
In the end, this commit fixed it: https://github.com/acados/acados/pull/1512/commits/3fb40e2034fb11d4a23280732ac4f2125ae366ef
But it is not really clear why.

giaf, 2025-05-02:
These structs are surely in they proper place now after the fix, alignment should be fine also in the wrong place provided that external fun workspaces keep 8-bytes alignment, as they seem to do.

It may be that some external function uses too much memory and so corrupts memory afterwards.
Did you identify the exact location of the segfault, i.e. the access to the invalid pointer?
And what happens if you put the structs back where they were (in the wrong place), and add a buffer in between the external fur workpsace and them? This means increasing the size of the ocp_nlp_workspace by xx bytes (with multiple of 8!), and advance the c_ptr pointer of the same amount before the ocp_qp_seed_assign.
If segfault disappears at a certain minimum value of xx in a fully reproducible way, it may be a good hint that the external fun workspace overwrite xx bytes afterwards.

giaf, 2025-05-02:
Alternatively, put after the external fun workspaces an array of some type of known value, as big as the seeds and res structs for example, and check that the array elements keep their value at the end of that seg-faulting example.