acados/acados #1522: ocp_nlp_out_set interface has changed.

HeiseCurtain, 2025-05-07:
I update the acados recently, and find the interface of c 'ocp_nlp_out_set' has changed from 6 params to 7. I wonder where can I get the detailed information about the update to guild me upgrading my code depend on this repo.  And I also wonder why to change the interface in the version of 0.5.0. Thanks very much.

FreyJo, 2025-05-07:
Hi,
the breaking changes are mentioned in the release notes:
https://github.com/acados/acados/releases/tag/v0.5.0
The changes were needed to ensure that the multipliers are always consistent with the `mask`, which are used to implement one sided constraints in acados, see https://github.com/acados/acados/pull/1333

closed 2025-05-09 by FreyJo

HeiseCurtain, 2025-05-11:
Thanks very much!!!