acados/acados #1529: Problems on running minimal_example_ocp.py

Coding-QQ, 2025-05-20:
I'm running acados on Windows, with python version 3.6.8. The project is based on this Python version so I cannot update it.

Now I just have two problems when running the example: minimal_example_ocp.py

First: A warning for object size: cados_solver_pendulum_ode.c:367:71: warning: argument 1 range [18446743627032952832, 18446744073709551408] exceeds maximum object size 9223372036854775807 [-Walloc-size-larger-than=]
  367 |     capsule->cost_y_fun = (external_function_external_param_casadi *) malloc(sizeof(external_function_external_param_casadi)*(N-1));

[image]

Second: It shows "__init__() got an unexpected keyword argument 'winmode"

[image]

Although I have checked and modified "shared_lib = DllLoader(shared_lib_name, mode=winmode)" Instead of "shared_lib = DllLoader(shared_lib_name, mode=winmode)" in file utils.py, then the error becomes:
OSError: [WinError 126] Could not find the specified module (I don't know whether I translate it correctly from Chinese to English)

[image]

Thank you very much if you could fix this

Coding-QQ, 2025-05-20:
By the way, my Matlab interface can successfully run the example "minimal_example_ocp.m"

FreyJo, 2025-05-22:
Sorry, the python interface requires Python 3.8 or later right now. If you have compatibility fixes for Windows and older python versions, I would consider merging them. But from the developer team, there is no interest to address this right now.

closed 2025-05-22 by FreyJo